pimusicbox/pimusicbox #203: AirPlay Drops Continuously

InsecureSpike, 2015-01-02:
I'm having so much trouble ever since the last update, with just AirPlay! it is very unreliable either using my iPhone or my Mac, radio streams work very well.
sometimes it'll drop while playing the first track other times it can make it maybe 4 tracks into an album!
does anyone have any ideas I can try?
thanks

jackrichardson, 2015-01-05:
You're not the only one - https://discuss.mopidy.com/t/musicbox-0-5-1-shairport-behaviour-strange/407/7
Until it's fixed I'm settling for pimusicbox 0.5.

woutervanwijk, 2015-01-13:
Since 0.5.1 shairport-sync is used in MusicBox. That probably is the problem. Maybe an update to that software fixes this.

teq0, 2015-01-13:
According to that mopidy thread the new version of shairport-sync does indeed fix the issue.

coreyk, 2015-01-18:
Building the newest shairport-sync from source on pi music box does indeed fix this issue.

woutervanwijk, 2015-01-19:
Fixed in 0.5.3 using the new version of shairport-sync

closed 2015-01-19 by woutervanwijk