pimusicbox/pimusicbox #209: Sleep timer

teq0, 2015-01-13:
This is a feature request - how about adding a sleep timer? I added one to the web client and it was pretty simple but it means having to leave the browser window open (the timer is in JavaScript), so I haven't bothered doing a PR.

I started to look at how/where to add it to the server side but didn't get very far. Would this require a change to mopidy, or is there a way to add it to Musicbox? I'd be happy to try to add it if you could point me in the right direction.

kingosticks, 2015-01-14:
Presumably you want to control this via a new/existing web page, in which case the simplest way might be to write your own http:app extension.  I think it'd also be possible to incorporate it into the existing musicbox-webclient by registering it as a http:app alongside the existing http:static extension. See https://docs.mopidy.com/en/latest/api/http-server/#tornado-application-example and take a look at how the mopidy-websettings extension works. 

Or you could try to build upon https://github.com/Zashas/mopidy-alarmclock/tree/master/mopidy_alarmclock. You could add your functionality (if it's missing) and maybe work on integrating it with musicbox-webclient in a similar way that musicbox-websettings does (or is at least supposed to - last time I looked this was bugged).

teq0, 2015-01-14:
Great, thanks, I'll check it out.

michaellunzer, 2015-02-11:
Hi @teq0 have you looked into any of the suggestions of @kingosticks suggestions? I'm interested in creating an alarm clock and sleep timer also. Maybe we can collaborate?

kingosticks, 2015-02-11:
An option for this is currently being discussed at https://github.com/mopidy/mopidy/issues/956

kingosticks, 2015-02-25:
I don't think an issue here is useful.

closed 2015-02-25 by kingosticks

teq0, 2015-02-26:
I'll be submitting the PR to mopidy in the next few days, I've already forked and updated Mopidy-MusicBox-Webclient, should I raise an issue there before I submit a PR? Or just submit the PR and discuss it then?

kingosticks, 2015-02-26:
In general, a PR that solves an issue may be more likely to get in as it
should have hopefully demonstrated it's usefulness. Don't forget your tests!

teq0, 2015-02-26:
I'm writing the python tests right now for the mopidy/core changes, but I'm not seeing any tests in the Mopidy-MusicBox-Webclient. Am I missing something?

kingosticks, 2015-02-26:
No. There's no tests there. User interface tests are a pain. But if they
are useful, feel free to add some.

michaellunzer, 2015-08-21:
Has the Pull Request been complete? Just wondering the current status

kingosticks, 2015-08-22:
Yeh, it'll be in the 1.0 release which will hopefully be out next week, or
thereabouts.

teq0, 2015-08-23:
Sorry, which PR? Mine (https://github.com/mopidy/mopidy/pull/1085) was closed without being merged (it's way out of date now anyway) because the discussion turned to having a generic extension capability for core.

I'm waiting to see how that pans out, if it happens the sleep timer will need a pretty major rewrite, I'm assuming.

kingosticks, 2015-08-24:
> Sorry, which PR? Mine (mopidy/mopidy#1085

Something got merged. I'll have to check back later on that one.

kingosticks, 2015-08-24:
Oh maybe I'm getting confused with the alarm clock thing.