pimusicbox/pimusicbox #211: Can't seem to get youtube playlists to work

rathboma, 2015-01-27:
Hey guys,

Firstly -- musicbox is fantastic! Very happy overall. Want to make that clear :-).

I'm having some problems playing youtube playlists. 
eg: https://www.youtube.com/playlist?list=PLDWrj6ynqDMGiREzFXF_JkuwBi3-pUKJH
I've tried two things:

1. Go to streams, choose 'youtube' enter youtube playlist uri, click play, nothing happens.
2. Go to search, choose 'youtube' enter youtube playlist uri, it finds the first song, but not the whole list.

I looked at the mopidy youtube connector (https://github.com/dz0ny/mopidy-youtube), and tried a generic stream prefixed with `yt:`, but that doesn't seem to work either.

Any thoughts? Am I just doing something stupid or have I uncovered a known issue?

Matthew

kingosticks, 2015-03-04:
If you have not done so, please first update mopidy-youtube to the latest version by running `pip install --upgrade mopidy-youtube`. If you are still experiencing problems please post at discuss.mopidy.com as this issue tracker is really meant for bugs and feature requests rather than user support. Thanks

closed 2015-03-04 by kingosticks

Travattack, 2016-12-07:
i did the upgrade. but now it seems that my webpage isn't working anymore. any idea?
i dit an apt-get upgrade after to updateeverything.
still nothing.

i got this;
 ERR_CONNECTION_REFUSED?

kingosticks, 2016-12-07:
Recent versions of mopidy-youtube probably depend on the later versions of Mopidy which Musicbox does not currently support. Musicbox v0.6 was not designed to be upgraded using apt-get.