pimusicbox/pimusicbox #212: Update Some Settings Without Reboot

artistro08, 2015-01-30:
it would be awesome if you could update some settings, (e.g: SSH, AirPlay, UPnP) without rebooting. it seems like a long shot but Just wondering if this could happen.

kingosticks, 2015-01-30:
That's a good idea. The pi can take a long time to reboot.

artistro08, 2015-02-03:
Wouter? is this possible? I also found out OpenELEC has this feature. Maybe we can get it from there or use their code somehow. It's embedded software though

kingosticks, 2015-02-03:
You just need to add some extra stuff to https://github.com/woutervanwijk/mopidy-websettings to catch those services that can be updated and do something like an `os.system('service x restart')` for them.

artistro08, 2015-02-03:
Ahh. it seems easy but, I honestly don't know much about programming xD I will check it out though

benjbubu, 2015-02-22:
Does the command /etc/init.d/mopidy restart working for you ? I edit some config in settings.ini and do this command and it works for me but don't know if it restarts all the service

kingosticks, 2015-02-22:
That'll work just fine for all Mopidy settings, but changing some of the other settings (e.g. enabling shairport/upnp or changing between RPi outputs) will require something extra.

artistro08, 2015-02-22:
yeah that's what I was thinking. Maybe output changing could still reboot and other settings such as ssh, upnp/dlna, airplay, library update etc

kingosticks, 2015-02-25:
https://github.com/mopidy/mopidy/issues/502 is relevant for the Mopidy settings

artistro08, 2015-02-26:
i see that. is there a way you can execute some .sh scripts for dropbear, shairport-sync, and upmpdcli to edit the init scripts and kill the processes over php or javascript?

artistro08, 2015-03-03:
I saw this. thought it might be of help to link through the cgi interface and restart some scripts using the init.d interface. don't know if this would help

www.webmin.com

kingosticks, 2015-03-04:
It's pretty trivial for us to run any arbitrary command on the local system so (re)starting services from within the startup.sh or mopidy-websettings is easy enough once we know the mapping of settings to services. 

We certainly want to fix startup.sh to run the minimum commands necessary to make the system match the current settings file. At the moment I think it redoes nearly everything each time it is run, regardless of what changed (there are some exceptions to this e.g. hostname). Then mopidy-websettings just needs to run the script. 

In an ideal world, Mopidy Core would include a config API that mopidy-websettings could use use to get the current config and submit changes, much like the one @adamcik envisions in mopidy/mopidy#502. Any changes could be saved to the user's config file (tricky when settings are amalgamated from multiple files) and events sent to parts of Mopidy that changed.  Then you'd need just a simple mopidy-musicbox extension which defined the musicbox specific config and listened for the the config changed events. On receiving an event it would then perform whatever extra service/system restart or external config change was necessary.

artistro08, 2015-03-04:
Ahh I see now. Thank you for explaining. I can definitely see this happening under these circumstances. i'm willing to test anything so i'll be on the lookout when a new version comes up