pimusicbox/pimusicbox #213: CIFS Problems with OS X 10.9 or 10.10

ellis2323, 2015-02-02:
At end of a file transfer, there is an error "The file is already used". The finder try some tricks in dot files and save informations in .DS_Store. The solution is easy to impletement: disallow .DS_Store. I use the veto files and delete veto files directives.

I recommend to modify smb.conf with this:

    [Music]
        path = /music/MusicBox
        comment = Directory for music files
        writable = yes
        browseable = yes
        guest ok = yes
        create mask = 0777
        directory mask = 0777
        veto files = /._*/.DS_Store/
        delete veto files = yes

Best regards,

jcass77, 2015-02-24:
Does mounting samba shared drives on Mac OS X work out of the box for you?

I had to add ```sec=ntlmssp,nounix,noperm,noserverino``` to startup.sh to get the directory to mount properly on my machine.

ellis2323, 2015-02-27:
It work out of the box but there is a '\" in the name of the share. It is 'musicbox\.local'.