pimusicbox/pimusicbox #214: Support for PulseAudio network streams

Barbers, 2015-02-03:
Hi,

I would like to see support for PulseAudio network streams on Pi-Musicbox.
The main goal is to use the Pi also as remote audio card for my pc.

Greetings
Barbers

LouisKottmann, 2015-12-03:
I have the same need

Even though pulseaudio-ndla works, it incurs several seconds of audio latency that makes watching movies with sound through musicbox impossible.

However, I know that using a pulseaudio sink you have little to no latency.

kingosticks, 2015-12-04:
Does shairport-sync support pulseaudio?

LouisKottmann, 2015-12-04:
Their README says "It will not work properly – if at all – with a PulseAudio (pseudo-)output device" but they do have a (short) `audio_pulse.c` source file

kingosticks, 2015-12-04:
I think their develop branch is a lot further along so that may not be true
anymore. Probably worth asking the Dev.

almereyda, 2018-11-02:
According to https://github.com/mikebrady/shairport-sync/search?q=pulseaudio&unscoped_q=pulseaudio this is called the `pa` backend.

This is present since the 3.1 release https://github.com/mikebrady/shairport-sync/releases/tag/3.1

Which is used by PiMusicBox https://github.com/pimusicbox/pimusicbox/blob/master/create_musicbox0.7.sh#L3

Which can be activated with https://github.com/mikebrady/shairport-sync/blob/master/scripts/shairport-sync.conf#L88-L93

Adding this to the file and restarting the service doesn't cause any errors. But the device also does not show up in `pavu-control` on a machine in the LAN when `pulseaudio-dlna` is running, while UPnP is activated in the router, and no firewall is active on the MusicBox.