pimusicbox/pimusicbox #216: Support Rasberry Pi 2

drewdotpro, 2015-02-04:
Works great on my Pi Model B.

Tried it with Pi 2 Model B, doesn't work =(

kingosticks, 2015-02-04:
You need to update the base raspbian install. Option 3 at
http://thepihut.com/blogs/raspberry-pi-tutorials/16982376-updating-raspian-on-your-microsd-for-the-raspberry-pi-2

koutloup, 2015-02-05:
And if you only have a Pi 2 Model?

kingosticks, 2015-02-05:
Then perhaps you could try option 2 but you'll have to hunt out what files
you need to update. Else you'll have to wait for an updated version of
musicbox. Hopefully there will be one soon.

kingosticks, 2015-02-05:
Or I think there's actually a script to create a musicbox image from a
fresh raspbian image. It was updated fairly recently so it might still
work... It's in the github repo.

koutloup, 2015-02-05:
Script sound promising, i will take a look at it.

kingosticks, 2015-02-10:
When you get a chance please report back how you get on and if the script needs any updates.

gonace, 2015-02-14:
I did the system upgrade as per the guide in the link posted in this thread, and now MusicBox crashes when trying to play a track/stream of any sort.

I've tried playing:
* A local track
* A track on a network share
* Spotify
* TuneIn

All results in a crash of MusicBox, I don't know what log to look in? I've looked in the following with no success.
/var/log/mopidy/ 
/var/log/monit.log

The only thing I could find is this (in: /var/log/monit.log)
    [CET Jan 15 10:55:17] error    : 'mopidy' process is not running
    [CET Jan 15 10:55:17] info     : 'mopidy' trying to restart
    [CET Jan 15 10:55:17] info     : 'mopidy' start: /etc/init.d/mopidy
    [CET Jan 15 10:57:17] info     : 'mopidy' process is running with pid 2596
    [CET Jan 15 10:09:22] error    : 'mopidy' process PID changed from 2596 to 2793
    [CET Jan 15 11:11:22] info     : 'mopidy' process PID has not changed since last cycle
    [CET Jan 15 11:39:06] info     : monit daemon with pid [2522] killed

kingosticks, 2015-02-14:
When you say it crashes, does the whole system become unresponsive (i.e
can't login via ssh or physically with attached keyboard) or is it just the
mpd/Web interfaces?

You should be able to observe more by logging in (via ssh or physically),
stopping monit (sudo service monit stop), stopping mopidy (sudo service
mopidy stop), running mopidy in the terminal with full logging (sudo -H
-u mopidy bash -c'/usr/local/bin/mopidy --config/etc/mopidy/mopidy.conf
-vvv').

I did the system upgrade as per the guide in the link posted in this
thread, and now MusicBox crashes when trying to play a track/stream of any
sort.

I've tried playing:

   - A local track
   - A track on a network share
   - Spotify
   - TuneIn

All results in a crash of MusicBox, I don't know what log to look in? I've
looked in the following with no success.
/var/log/mopidy/
/var/log/monit.log

The only thing I could find is this (in: /var/log/monit.log)
[CET Jan 15 10:55:17] error : 'mopidy' process is not running
[CET Jan 15 10:55:17] info : 'mopidy' trying to restart
[CET Jan 15 10:55:17] info : 'mopidy' start: /etc/init.d/mopidy
[CET Jan 15 10:57:17] info : 'mopidy' process is running with pid 2596
[CET Jan 15 10:09:22] error : 'mopidy' process PID changed from 2596 to 2793
[CET Jan 15 11:11:22] info : 'mopidy' process PID has not changed since
last cycle
[CET Jan 15 11:39:06] info : monit daemon with pid [2522] killed

kingosticks, 2015-02-14:
Also, what type of audio output are you using?

gonace, 2015-02-14:
I'm using the analog audio output, the error I'm receiving is the following:
*** glibc detected *** /usr/bin/python: munmap_chunk(): invalid pointer: 0x76da5b18 ***
Aborted

PierrickMartos, 2015-02-15:
Same as @gonace musicbox works fine on raspberry pi but when i did the system upgrade to raspberry pi 2 and try to play music, result in a crash

kingosticks, 2015-02-15:
I can't think why that problem would arise. Weird. There is also the option
of creating a musicbox image from scratch using the script in github.
Hopefully @woutervanwijk will be able to upload a rpi2 compatible image
soon.

woutervanwijk, 2015-02-16:
I am not able to work on MB atm, sorry. 

From what I heard, there seem to be two problems running MB on the Pi 2: upgrading the system and the detection of the soundcard. Problem one is solved quite easily by using an older PI, a microSD and a SD converter card. Install MB on the card, run it on the old Pi and do apt-get dist-upgrade. Then put the mircoSD in the Pi 2. It will boot. But the setsound.sh script ( https://github.com/woutervanwijk/Pi-MusicBox/blob/master/filechanges/opt/musicbox/setsound.sh ) is not correctly detecting the analog soundcard of the Pi 2, so the script breaks, as does the sound setup... 

I will try to update it if I can work on the project again...

tovabbi, 2015-02-20:
I would be great if someone could provide an image of a dist-upgraded system as suggested by Wouter. I have only a Pi 2 so I cannot do the trick. A fork would be even nicer.

kingosticks, 2015-02-20:
I think we had a couple of reports on discuss of people having problems
with the update method. I need to get another microsd card to try this
myself.
The 2nd problem I hadn't heard about.

If you are unable to work on it right now, would you be open to adding me
(or someone) as collaborator so we can keep things ticking over?

jorisbaiutti, 2015-02-21:
Same Problem with this Manual: https://github.com/woutervanwijk/Pi-MusicBox/blob/master/Create%20Pi%20MusicBox.rst  It crashes when i play a song:( I hope there will be a update soon.

kingosticks, 2015-02-21:
To clarify, you took a fresh recent raspbian image and followed those steps
and it still crashes? That doesn't sound good.

woutervanwijk, 2015-02-22:
It probably crashes because the soundcard/alsa is not configured properly. Does it still crash if you manually set /etc/asound.conf to something like (for the analog card): 

pcm.!default {
    type hw
    card 0
}
ctl.!default {
    type hw
    card 0
}

woutervanwijk, 2015-02-22:
@kingosticks I added you as a contributor to both Pi MusicBox and the Webclient!

kingosticks, 2015-02-22:
Thanks very much @woutervanwijk. l see if I can make a dent in some of these issues, starting with this one.

jorisbaiutti, 2015-02-23:
Thank you very much!!!! @woutervanwijk It works :) But every Time I restart my Raspberry the settings in the asound.conf file are gone.

kingosticks, 2015-02-24:
The setsound script runs on each reboot and it overwrites that file. Fancy testing the updated version?
`cd /opt/musicbox
wget https://raw.githubusercontent.com/woutervanwijk/Pi-MusicBox/master/filechanges/opt/musicbox/setsound.sh

kingosticks, 2015-02-24:
Well, this is a bit annoying. Seems since raspbian moved to a 3.18.x kernel the way the sound drivers needs to be loaded has changed. It seems it needs to be hardcoded into config.txt which means we actually need more changes here.

kingosticks, 2015-02-25:
What is now checked in should work for analogue, hdmi and usb soundcards. It probably won't work for anything i2s based.

jo-me, 2015-02-25:
Hey kingosticks, 
I upgraded my musicbox installation but kept the SD card in the Raspberry B (1). 
The behavior as described above is the same. Mopidy is crashing with:
*** glibc detected *** /usr/bin/python: munmap_chunk(): invalid pointer: 0xb6e08b18 ***
whenever something was starting to get played.

I updated the setsound script with the latest version from git and the error disappeared! 
I'm running Musicbox on a Raspberry B (1) using analog output.

Note to others updating the script: 
In my case the wget command from above downloaded the file but named it differently ( ".1" at the end). Don't know why, maybe it was accessed by something. 
So, I removed the original file first, downloaded the new file, Then did a "chmod +x setsound.sh" to make it executable. After a reboot (or manual execution of the script and the mopidy service) it worked.

Thanks a lot
Jochen

kingosticks, 2015-02-25:
That's great, thank for the update and for explaining those extra commands
that are necessary. I hopefully should be able to test with an i2s card
tonight, there's an update to /boot/config.txt that should fix those (for
anyone who wants to try).

Stealthii, 2015-02-25:
I can confirm that fixes the issues noted above (other issues still prelevant)

Using Pi Model B Rev2 with Hifiberry DAC

kingosticks, 2015-02-25:
Which issues are now left outstanding? I think i2s cards are now sorted.

kingosticks, 2015-02-25:
For reference, I've had success with both option 2 and option 3 at http://thepihut.com/blogs/raspberry-pi-tutorials/16982376-updating-raspian-on-your-microsd-for-the-raspberry-pi-2

For option 3, you need to update the two files as mentioned in my previous comment.

For option 2, I downloaded and mounted the latest raspbian image and replaced the contents of musicbox's boot partition with the raspbian one. I then copied the 4 files under https://github.com/woutervanwijk/Pi-MusicBox/tree/master/filechanges/boot back into the musicbox boot partition.

drewdotpro, 2015-02-26:
Once this is "Raspberry Pi 2 compatible" will you be able to release a flashable image, like @woutervanwijk did with the first Pi?

kingosticks, 2015-02-26:
I think this is ready to go, I've exhausted the testing I can do on my
devices.

drewdotpro, 2015-02-26:
Amazing, and awesome. Thank you so much. I very much look forward to putting an image file on my Pi 2 =D Just lemme know where I can download from!

woutervanwijk, 2015-02-26:
From here, thanks to the work of @kingosticks http://www.pimusicbox.com/musicbox0.5.4beta.zip

jodal, 2015-02-26:
RPi2 support is maybe a good opportunity to bump the version number to 0.6?

EinArzt, 2015-02-26:
Is the image only for RPi2 or can it also run on a RPi B+?

kingosticks, 2015-02-26:
It should work fine on all models.

jodal, 2015-02-26:
As usual, I've mirrored the release at http://dl.mopidy.com/musicbox0.5.4beta.zip.

Come to think of it, why not test hosting the images with the GitHub release feature?

> At this time, we do not place any limits on the size of binary uploads nor the bandwidth used to deliver them.
-- https://help.github.com/articles/about-releases/

Phocee, 2015-02-27:
Downloading. This is great, thanks for your work !

xiolo, 2015-02-27:
Tried the 0.5.4 img on my Pi2, but it wasn't working with USB sound cards (tried 2 different USB sound cards that are loading OK at  boot time based on dmsg messages and listed ok under amixer). Ended up using setsound.sh from a previous 0.5.3 release and USB sound output works ok now.

Any idea ?

kingosticks, 2015-02-27:
Can you provide the output of `aplay -l`

PierrickMartos, 2015-02-27:
Thanks, the img works fine on RPi 2!

celeryclub, 2015-02-27:
I can verify that this image works correctly on a RPi B+ using i2s output.

xiolo, 2015-02-27:
On the working 0.5.4 beta PI 2 (with setsound.sh from 0.5.3)

aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: DAC [USB Audio DAC], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

kingosticks, 2015-02-27:
There is a fixed version at https://raw.githubusercontent.com/woutervanwijk/Pi-MusicBox/master/filechanges/opt/musicbox/setsound.sh if you want to try it.

artistro08, 2015-02-27:
can confirm that the script above works awesome!

xiolo, 2015-02-28:
Unfortunately, not for me. Reverting back to version 0.53 of setsound.sh

Here is the output of aplay -l when "not working":

aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: DAC [USB Audio DAC], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

kingosticks, 2015-02-28:
Thanks @xiolo.  I've fixed my dumb mistake now and found a better way to test it. I think this will work (famous last words).

xiolo, 2015-03-01:
yep... that fixed it  (USB DAC) with latest git version of setsound.sh. nice one :-)

kingosticks, 2015-03-01:
Super, thanks for your help.

artistro08, 2015-03-01:
Just updated as well. it works like a charm

UnlimitedpublicFriction, 2015-03-01:
i hope you don´t mind. my question is about google all access.
i see all albums, artists, radios on my google account.
if i select one, it is then under queue, but no audio output.
everything else, like tunein is working fine.
for testing i use the analog output.
(have not testet my usb dac with the new setsound.sh)
do i have to change anything (setsound.sh,mopidy.conf)?
regards
p.s.
aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

kingosticks, 2015-03-01:
If you are getting audio playback with the other services then it must be
something specific to google all access rather than raspberry pi 2 support.
I think we'd be best off discussing that in a new issue rather than using
this one. I'd be tempted to raise the issue with their project directly or
on discuss.mopidy.com first where you are more likely to hear back from
google music users (I don't use it myself).

Phocee, 2015-03-03:
Hello,
I've tried tonight using the RPI 2 and an I2S DAC (I-Sabre DAC ES9023). It doesn't work when I leave the M.B. Audio config param at "Automatic". The other choices are for Hifiberry and IQAudio. Could you please tell me what setup is required ? Thanks a lot.

kingosticks, 2015-03-03:
I'm not familiar with that card. The comment in the settings file states
that i2s cards cannot be auto detected, this is because we must first load
the correct kernel module for the card. Does your card require a particular
kernel module to function? Are you using the very latest version of
setsound.sh from github? I think if you use the very latest version of the
script and you don't need to load any modules, you should be OK to leave it
as automatic.

celeryclub, 2015-03-03:
I have an RPi 2 with an ES9023 i2s card running this image. It won't auto-detect the card, but setting it to "HiFiBerry" works perfectly.

Phocee, 2015-03-04:
I had to try harder: I had tried Hifiberry DAC+ only. It works indeed with HfiBerry DAC.
Thanks again !

twachti, 2015-03-09:
Is it possible to integrate HifiBerry AMP+ support to setsound.sh, or does it already work?

kingosticks, 2015-03-09:
Assuming the amp driver is in the raspbian kernel that was used for the latest release (probably was) then if you get the very latest version of setsound (i.e. the one I just committed) then it should work. Set the musicbox `output` option to 'hifiberry_amp`. 

If it doesn't work, please paste the result of running `aplay -l`.

twachti, 2015-03-11:
Thank you, it works, but it's to loud 😞
There is no problem if I use AirPlay, but via the webinterface 3/100 is loud as 50% via AirPlay.

artistro08, 2015-03-13:
i've noticed this too.. idk why but can it be fixed? May i start a new Issue on this?

[2 more replies]

closed 2015-03-13 by kingosticks