pimusicbox/pimusicbox #217: Special characters in spotify password not working

af2005, 2015-02-07:
The spotify password ist not allowed to have special characters. I had a password containing a question mark and a comma. 
The log file of Mopidy said 'incorrect login/password' although correctly configured.

Guess the comma was the problem but I haven't tried with only a question mark. 
Might need some " " around the password string or whatever. 

Great piece of software btw!

pattyland, 2015-05-19:
FYI: Putting double quotes around username and password didn't fix the problem for me

kingosticks, 2015-05-19:
There is a PR that fixes this but it needs some love. I've been putting it
off but it needs to be done. Will look again. I think there is probably a
way around this by specifying (just) your Spotify config in one of the
other config files that mopidy automatically reads in. Maybe the one in
/root/.config?