pimusicbox/pimusicbox #218: spotify family

napoleonmm83, 2015-02-08:
Id would be usefull to setup many spotify accounts for different members in the family.
So my son like a differnt music style than my and i dont want to mix up my play list with him music.

kingosticks, 2015-02-24:
Fundamentally this isn't possible without restarting the Mopidy service every time you want to switch profiles. If you had anything more than a few playlists in your account it would be quite slow to switch users on a raspberry pi, during which time the web interface would also be down. I don't think there is a currently a reasonable way to implement this so I'm afraid it's not going to happen. If someone wants to come up with something we can revisit it then.

closed 2015-02-24 by kingosticks

jodal, 2015-02-24:
Mopidy-Spotify is using libspotify for interfacing with Spotify, and according to libspotify's docs it uses some global state, making it impossible to have multiple libspotify sessions in the same process. Each libspotify session is tightly coupled to a Spotify username and password. Thus, as @kingosticks points out, just about the only feasible solution is to restart the Mopidy-Spotify backend to switch users.