pimusicbox/pimusicbox #219: Fallback to adhoc wifi?

adamcik, 2015-02-10:
See https://discuss.mopidy.com/t/start-ad-hoc-network-if-standard-wlan-is-not-available/538 for original request.

mattbrailsford, 2015-02-14:
What would people want the adhoc network to do? The way I see it, there are two options.

1) It runs standalone only playing local files
2) It runs a config webpage whereby users could configure settings of the musicbox.

Personally, I'd love to see option two as once you embed your Pi in a device, there is currently no easy way to reconfigure it if you either change any login details for any plugins, or connect to a different wifi connection.

If going for option two then, it would be great if we could implement a simple web server to host a page to manage musicbox settings. Also, having this trigger from the status of a pin would also be handy, ie, being able to switch the Pi into config mode manually, not just when the wifi doesn't connect.

What would others want from having musicbox fallback to an adhoc network?

kingosticks, 2015-02-14:
Option 2 sounds very useful. But is the musicbox settings page not
sufficient?

mattbrailsford, 2015-02-14:
I totally forgot about that. Yes, yes it would :)

kingosticks, 2015-02-24:
A PR implementing this would be very welcome.

zenonasz, 2015-04-14:
I have seen some code in /etc/network/if-pre-up.d/linux-wlan-ng-pre-up.

Do we know the status of this feature?

Regards,
Zenon

zenonasz, 2015-05-30:
Any news on this?
Regards

kingosticks, 2017-10-20:
Wifi Direct would be cool too: https://raspberrypi.stackexchange.com/questions/45451/connect-to-raspberry-pi-3-over-wi-fi-direct