pimusicbox/pimusicbox #220: Spotify browsing gives "No tracks found..."

liftarn, 2015-02-15:
I go in Browsing -> Spotify Browse and select, for instance, Featured playlists. I click on any of them and all I get is "No tracks found..."

kingosticks, 2015-02-24:
Is this still an issue for you? Is this with default webclient (mopidy-musicbox-webclient) ? Do regular Spotify playlists work?

liftarn, 2015-02-24:
I'm using Pi MusicBox 0.5.3.
I get no Spotify playlists in the playlist tab.

kingosticks, 2015-02-24:
So this is a general Spotify issue rather something isolated browsing. Do
you have a Spotify premium account? Are there any errors in the logs at
/var/log/mopidy regarding Spotify login?

liftarn, 2015-03-13:
It is a premium account. No visible error messages in the web client. Will try to extract a log.

closed 2015-08-31 by jodal