pimusicbox/pimusicbox #222: autoplay = last played (enhancement)

jasonhurd, 2015-02-18:
I would like to request the option to automatically set the autoplay url to the currently playing stream or "last played".

Or have options on the "Now Playing" page to set this stream to autoplay url.

Would also be a good idea to have the option on the "Now Playing" page to "save this stream to Streams List" or Favorites. etc.

Thanks,

Jason

kingosticks, 2015-03-05:
An option to set the autoplay URL to what's currently playing song can be added to websettings, that's simple.  However, last played is trickier (until Mopidy v0.20, at least) and even then I'm not sure it's very useful; how often do people really change the autoplay setting?. 

Setting autoplay from within the "Now Playing" page would require musicbox-webclient to know about Musicbox's autoplay functionality. It does already know about Musicbox at a high level when deciding if it should display the settings and system navigation links but I'm unsure if we want to couple it so tightly. Does the usefulness really outweigh the added complexity?

There is already a button to add the currently playing stream to the saved streams on the "Streams" page, but I agree it'd be nice to also add a button on the "Now Playing" page.

jasonhurd, 2015-03-06:
Thank you for the consideration. I understand and agree with not wanting to convolute the project.

I would be very happy to have a simple solution such as a button to set the autoplay URL to the currently playing. This solution would at least streamline setting up the new autoplay.

Thank you again,

Jason