pimusicbox/pimusicbox #223: every device has it own profile

anielk, 2015-02-19:
Hi,
I use different kinds of computer/laptops/tablets. 
And on each device it has it own settings. so I need to change it on every device.
so on my laptop I have the streams I have saved, but that is only on my laptop. On my desktop computer everything is default.

Thanks
Aniel

kingosticks, 2015-02-19:
If you want the ability to access saved streams across multiple devices
then there is at least one topic at mopidy.dicuss.com addressing that
issue. If you want something else then you'll have to be a lot more
specific.

kingosticks, 2015-02-19:
This is the discuss thread
https://discuss.mopidy.com/t/possibility-to-store-dirble-tunein-radio-channels-on-sd-card/415

anielk, 2015-02-20:
sorry for the incomplete information I gave.
What I meant is:
The musicbox is on IP i.e 192.168.1.10 (I have only one musicbox)
And if I connect with my laptop to the musicbox I see all the streamstation I saved.
But when I connect with me android phone to the musicbox and I want to pick a different stream from my stream list. Then they are not there. All I see is the default streams. So after changing again all the streams I connect to the musicbox with my Ipad and again all I see is the default stream.

later I gave the musicbox a DNS.
If I connect with my laptop via IP then I see me saved streams. 
But when I connect via DNS to musicbox.blabla.local then all my stream are gone.
Is there a way to have only 1 profile or something so I will always see my saved settings.
Hopefully the problem is a bit clear now?

I will have a look at the link you provided 

thanks again
Aniel

kingosticks, 2015-02-20:
Streams that are saved using the browser are stored in the device's local
browser data. If you want to see the same stream list on all devices then
you must edit the file on the musicbox server. I'm  link also provides some
other ways of achieving a similar goal.

kingosticks, 2015-02-24:
Did this solve your issue?

anielk, 2015-02-24:
Yes thanks!

closed 2015-02-24 by anielk