pimusicbox/pimusicbox #224: No audio/streaming after network drop 0.5.3

jasonhurd, 2015-02-20:
While testing the PiMusicBox I have found that after the network drop outs (unplugged cable, etc) for more then a few minutes it does not resume playing after a network connection is reestablished. It does appear to reestablish the stream if the disconnection is only for a short time.

Navigating to the web client 'playing now', you can reinitialize play by pausing then playing.

jackrichardson, 2015-03-08:
This is not a new issue: https://github.com/mopidy/mopidy/issues/720
It can be rectified by installing mpd-watchdog. Instructions are available here: https://discuss.mopidy.com/t/radio-doesnt-return-after-internet-dropout-mpd-watchdog/424/8
@woutervanwijk Seeing as it is such a small package, could it please be added to the base pi-musicbox image?

jasonhurd, 2015-03-09:
Thank you for your reply. I have installed the mpd-watchdog and appears to work well.

I did however have to manually create the tkem.list file as the commands did not seem to want to create it for me.

Thanks again!

jackrichardson, 2015-04-01:
Any chance mpd-watchdog could be included in Pi Musicbox 0.6?

jasonhurd, 2015-07-16:
Hello All

I continue to have issues in this area. I'll give a bit of background:

I have four of these setup at my boss' place (two raspi B running v0.53, and two raspi 2 running v0.6). All devices have the watchdog installed. The main purpose is to continuously stream four local AM radio stations throughout the house via a distributed audio system, over the air reception is very lacking, so that when in any room can access any one of the four via the local control box.

You may ask, why four of these as you can change the stream easy enough? Well, it boils down to convenience for him. To stick with what he is used to, mainly, because as far as he is concerned nothing should have changed except the audio quality.

Anyway, now to discuss the issues. Something happens where the streams all drop out. I had assumed this was interruptions in internet connectivity (and may still be the case) but it is hard to confirm since I don't live at the residence. All radios have the watchdog installed and after testing this service by pulling the network cables out of all the Pis, then plugging them back in after a minute or so, the streams resume playing. By time the boss gets home though, all streams have stopped. When I head up to the house the next day to check it out, sure enough all streams have stopped and internet is active.

I am really pulling my hair out with this one. It dumbfounds me that this could be happening with four different units so it can't just be a random glitch. I've tried updating all units and I have played around with the wait time (from default 60s to 120s etc), but I can't seem to get anything to work.

Any suggestions?

Thanks for the help.

Jason

kingosticks, 2015-07-23:
You will need to check the logs and probably enable some more verbose
logging to make any headway with debugging a remote system like this. I
suppose you could try to set a local file playing on repeat to rule out a
non-network issue. I'd suggest trying some other stations incase it's
related to the speciifc streams (some sort of max connection time?).

jasonhurd, 2015-07-29:
Very good suggestions! I haven't even thought about the possibility of a max connection time.

I will setup a local test to see if I can get to the bottom of this. I shall report back.

Thanks again.

kingosticks, 2017-03-26:
Oh wow, I forgot about this one. Adding this to v0.7.0

jasonhurd, 2017-05-22:
Thanks @kingosticks !!!

commit 607d48e1 (2017-07-06): Add mpd-watchdog (Fixes #224).
closed 2017-07-06 by kingosticks