pimusicbox/pimusicbox #226: Can't connect to subsonic server

benjbubu, 2015-02-22:
When i configure my subsonic server into pimusicbox, the log give me a Error authentication. I try several things : different user, url in xxx.subsonic.org and the direct IP address but nothing works.

Any idea ?

kingosticks, 2015-03-09:
Could you try posting this at discuss.mopidy.com if you are still having problems.

benjbubu, 2015-03-09:
I just found the solution myself. I posted it here : 

Each version of Subsonic use a specific API version. You need to edit the file connection.py in /usr/local/lib/python2.7/dist-packages/libsonic and change the version of the API a tthe line "API_VERSION" To find your API Version, go to http://www.subsonic.org/pages/api.jsp

kingosticks, 2015-03-09:
So, there is scope here to at least improve our documentation.

kingosticks, 2015-03-09:
And is that really the only way to modify the API version? That is horrible

kingosticks, 2015-03-09:
The libsonic `Connection` class constructor accepts an `apiVersion` parameter which mopidy_subsonic isn't specifying. Perhaps it should be configurable.

@benjbubu What version of the server are you running? Does everything work once you change the version?

benjbubu, 2015-03-09:
I using the 4.8 version of Subsonic and the api version by default in libsonic is for 4.9. Anyway the last version of Subsonic is 5.2.1 and the number of api version is different. ANd yest everything is working fine !! :D

kingosticks, 2015-03-09:
Isn't the default api version v1.11.0 AKA subsonic version 5.1 (https://github.com/crustymonkey/py-sonic/blob/master/libsonic/connection.py#L25) ?

kingosticks, 2015-03-09:
Ahh ok, the released version is 4.9 (the bump to 1.11.0 only happened in the code repository yesterday!)

osos, 2015-05-11:
I cant seem to get it working with subsonic 5.2.1 even if changing the API_VERSION to 1.12.0

Did anyone manage get it working on that version?