pimusicbox/pimusicbox #227: Idea : Stream playlist of another user in spotify ?

benjbubu, 2015-02-22:
Sometimes, spotify's user shares link of playlist in this form : http://open.spotify.com/user/nameofuser/playlist/XXXXXXXXX. But i can't find how to open it in pimusicbox. I try to enter the URL directly in the "stream menu" and select spotify but it only add the XXXXXXXX in the queue and nothing happened.

kingosticks, 2015-02-22:
You need to convert that into a Spotify URI by replacing `http://open.spotify.com` with `spotify://` and replacing all `/` characters with `:` characters i.e. `spotify://user:nameofuser:playlist:XXXXXXXXX`

I think there is a future plan for it to understand the `http://open.spotify.com` links but we are not there yet.

benjbubu, 2015-02-22:
Ok thanks man for the tip :)

closed 2015-02-22 by benjbubu

jodal, 2015-02-22:
The way ahead looks like this:

- pyspotify 2 already understands open.spotify.com and play.spotify.com links.
- Mopidy-Spotify 2 which use pyspotify 2 is soon to be released.
- Mopidy issue #696 will hopefully be implemented in v0.20, making it possible for Mopidy-Spotify to claim ownership of open.spotify.com and play.spotify.com URIs.

When this is done, Pi Musicbox must upgrade to the new Mopidy-Spotify and Mopidy versions.

tigerdesignsolutions, 2018-01-12:
Sorry to revive this, but you seem to know what you're talking about @benjbubu 

Is there a way to put in the URI to turn on the repeat and/or random?

benjbubu, 2018-01-14:
Mmm that is an excellent question. I dont know yet. We need to check if this kind of URL exists in spotify and see if it could be port in pimusicbox. Honnestly i didnt check the code of this soft for a while, i need to refocus on it