pimusicbox/pimusicbox #228: Authenticate Mopidy with music services. Error: 401 - Unauthorized

mvaneijgen, 2015-02-23:
When going to https://www.mopidy.com/authenticate/# and hitting the big orange button "Connect with Soundcloud" I get the error "Error: 401 - Unauthorized"

jodal, 2015-02-23:
Works for me.

Where do you see the error message? In the main browser window or in the popup?

mvaneijgen, 2015-02-23:
In a popup 
https://www.dropbox.com/s/y26aabt6ayqs9am/modpi.mp4?dl=0

jodal, 2015-02-23:
It went at bit fast at the end... after clicking the Facebook login button, did you click "connect me" too, or did it go directly to the 401 error?

mvaneijgen, 2015-02-23:
I Hit the 'Connect' button and after that got the error

jodal, 2015-02-23:
Can you reproduce this in other browsers than Safari?

mvaneijgen, 2015-02-23:
No I can not. Chrome works.

jodal, 2015-02-23:
I don't have Safari available myself right this moment, but I'll hopefully have a look at this later. I've opened mopidy/mopidy-website#2 to track this issue. Feel free to close this one (I can't).

closed 2015-02-23 by mvaneijgen