pimusicbox/pimusicbox #231: Typos and outdated references in various files and scripts

jcass77, 2015-02-24:
I've created pull request https://github.com/woutervanwijk/Pi-MusicBox/pull/230 to fix a few small potential issues mainly in the create_musicbox.sh script:  

- chown' and 'chmod' references are outdated.

- create '/var/lib/mopidy/.config/mopidy' directory before symlink to it

- add '-fn' option to symlinks to allow overwrite of existing links (useful when re-running scripts).

- duplicate 'cifs-utils' in apt-get install routine.

kingosticks, 2015-02-24:
it might be quite nice to do away with 'Create Pi MusicBox.rst' entirely. Duplicating that information is unnecessary and any important info should be in the script's comments.

jcass77, 2015-02-28:
I've thought about this as well but 'Create Pi MusicBox.rst' does contain some useful information not covered in the script. For example: how to build the image from scratch using a clean Raspbian install, as well as some optional performance optimisations to consider in post-installation.

Perhaps we can just remove the overlapping sections from 'Create Pi MusicBox.rst' and replace it with an instruction to run ```create_musicbox.sh``` instead?

kingosticks, 2015-02-28:
Yep that's a better idea. In the long run if we can build up some accurate
docs to replace the now outdated user manual I think that'll be good. And
starting with creating the image seems as good a place as any.

jcass77, 2016-02-22:
I'm happy that most of this was addressed when https://github.com/pimusicbox/pimusicbox/pull/230 was merged.

closed 2016-02-22 by jcass77