pimusicbox/pimusicbox #232: Mopidy Randomly crashing on RPi 2

artistro08, 2015-02-27:
After updating to the new kernel, and adding all the things i like, Mopidy crashes. i have a log here. I can't seem to find the reason why. Log: https://dl.dropboxusercontent.com/u/2986273/mopidy.log

artistro08, 2015-02-27:
it seems that gstreamer is crashing it. any way of fixing this?

heres another log: https://dl.dropboxusercontent.com/u/2986273/mopidy2.log

kingosticks, 2015-02-27:
That first log is way way too big to view on my phone and the 2nd log shows
no crash. May not be related (since I can't see yet) but you've got that
scanning an awful lot of non-music files. If nothing else, that's a big
waste of time.

kingosticks, 2015-02-27:
And can I just confirm that when you say you 'updated to the latest kernel'
do you mean you manually updated the kernel with 'apt-get dist-upgrade' or
'rpi-update' from an old musicbox image or the very latest rpi2 compatible
musicbox image? Or do you just mean you used the latest rpi2 compatible
musicbox image?

artistro08, 2015-02-27:
I installed Pi MusicBox on a RPi B+, did a 'apt-get update' and 'apt-get dist-upgrade', set up the pi on the MusicBox Side, and then switched the card to the new RPi2

artistro08, 2015-02-27:
That was because I moved the filesystem to a 3TB hard drive and didn't delete the old partition on the SD card after. the second log shows this: `2015-02-26 22:24:37,768 - ERROR    GStreamer error: state change failed and some element failed to post a proper error message with the reason for the failure. Debug message: gstplaysink.c(1899): gen_audio_chain (): /GstPlayBin2:playbin20/GstPlaySink:playsink0`

artistro08, 2015-02-27:
and that was about the time it crashed as well. It cut off and cut back on. I've installed it twice to make sure.

kingosticks, 2015-02-27:
So are you using the latest version of musicbox or an older one?

kingosticks, 2015-02-27:
With the latest version what you describe should have been fine. With an old version you need to update a couple of files, and that's all detailed in https://github.com/woutervanwijk/Pi-MusicBox/issues/216

artistro08, 2015-02-27:
0.5.3 is the one that I'm using. last time I tried to install it, it wouldn't work on the SD card. I updated it from Option 3 in that thread you linked me to. I made sure that everything was updated and switched it over. then, after a while, mopidy begins to crash and restart. Sometimes it doesn't restart either. it gave me this error when i tried to start it again

```
(process:16590): GLib-GObject-WARNING **: type name 'GdlaG\' contains invalid characters
[Invalid UTF-8] **
GLib-GObject:ERROR:/@sr\x0b@rc/packages)BUA:120:_g_elAm_\xf4\xffpes_init: assertion failed: (t\x01pe\x08)
Aborted
```

artistro08, 2015-02-27:
after restarting, Mopidy works fine after with the exception of randomly restarting.

kingosticks, 2015-02-27:
0.5.4 is the latest version.  I've never seen that error before, doesn't look healthy. My only guess is the SD card is somehow corrupt, but either way my only suggestion at this point is to do a fresh reinstall with the new image. It's a pain but once you've done the image you only need to copy over your settings.ini file.

artistro08, 2015-02-27:
That explains a lot! Where can I download. Its not up on the website.

artistro08, 2015-02-27:
Just found it under the beta! Thank you

closed 2015-02-27 by artistro08