pimusicbox/pimusicbox #235: Raspberry Pi 2 Loading Issues #2

artistro08, 2015-03-01:
It's still crashing on my Pi 2B.

After crashing, I ran `service mopidy run -v` and got this message:

```
Traceback (most recent call last):
  File "/usr/local/bin/mopidy", line 9, in <module>
    load_entry_point('Mopidy==0.19.5', 'console_scripts', 'mopidy')()
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 521, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2632, in load_entry_point
    return ep.load()
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2312, in load
    return self.resolve()
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2318, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python2.7/dist-packages/mopidy/__main__.py", line 28, in <module>
    from mopidy import commands, config as config_lib, ext
  File "/usr/local/lib/python2.7/dist-packages/mopidy/commands.py", line 14, in <module>
    from mopidy.audio import Audio
  File "/usr/local/lib/python2.7/dist-packages/mopidy/audio/__init__.py", line 4, in <module>
    from .actor import Audio
  File "/usr/local/lib/python2.7/dist-packages/mopidy/audio/actor.py", line 9, in <module>
    import gst  # noqa
  File "/usr/lib/python2.7/dist-packages/gst-0.10/gst/__init__.py", line 193, in <module>
    from _gst import *
ImportError: /usr/lib/arm-linux-gnueabihf/libxml2.so.2: undefined symbol: xmlDefaultBufferSize, version LIBXML2_2.4.30
```
I should add that I have the filesystem moved to to a 3TB Hard Drive and its overclocked to:

```
arm_freq=1000
core_freq=500
sdram_freq=500
over_voltage=2

```

kingosticks, 2015-03-01:
Sorry but if overclocking and moving the root filesystem you are into
unknown territory. You could try asking that question on discuss but I
can't look into that myself. But I would hope that you've addresses the
thermal issues when overclocking as the pi2 runs hotter.

artistro08, 2015-03-01:
Thank you. I figured. It was overclocking. Don't know why it would since I overclocked the last one

closed 2015-03-01 by artistro08

kingosticks, 2015-03-01:
It's got a new chip in it with 4x the cores, all running faster. Plus it
was reportedly signed off at 800 MHz but factory 'overclocked' to 900 MHz.
It is going to run much hotter. You could try a heat sink or a smaller
overclock (personally I wouldn't bother).

artistro08, 2015-03-01:
Yeah, that's what I did. The overclock was waaaay to high. everything else works fine but mopidy corrupts itself as it seems. been running it for 8 hours straight with no problems on the default clock with the filesystem moved. plus the little speed it did offer wasn't worth it.