pimusicbox/pimusicbox #236: Make default image slightly smaller

hadess, 2015-03-01:
The current image is 1,022,361,600 bytes. There's 2 partitions in the image that contain data, and free space before (4,194,304 bytes) and after (15,728,640 bytes), so 1,002,438,656 of data.

Removing the free space would make it fit on my 1GB (1,015,808,000 bytes) SD card ;)

jcass77, 2015-03-06:
Making the current image fit onto a 1Gb SD card may be possible, and maybe we need to decide if it is one of the PM design goals to keep within this limitation?

There are a few more extensions that we can consider adding in the future (e.g. local-images, pandora support) which may eventually increase the size of the image well beyond 1Gb. At the very least, I assume that Raspbian itself will keep on increasing in size as time goes by making this even more difficult to achieve.

kingosticks, 2015-03-06:
If we were able to deliver a 1,002,438,656 Byte image that's going to leave you with less than 13 MB of free space, is that actually usable? Remember you will need space for logs, cached data and possibly other stuff. I'd probably sooner sell the 1 GB card to a museum and buy a truck-load of 8 GB cards with the profits!

hadess, 2015-03-06:
I could actually see benefits to having the OS be under 1GB, and logs, cached data, etc. be either on a separate partition on the same device, or on an external device (such as a USB drive).

It would make it easier to upgrade MusicBox without touching existing data, and, until I find a museum interested in such a device, would allow me to reuse hardware I have, rather than fill landfills with still functioning hardware.

hadess, 2015-03-06:
Also, given that the default configuration will resize the root on boot, the only people impacted will be those who, like me, choose to run the distribution off a small storage device.

I'd be happy reviewing the packages installed in the default image, see if we can trim the default installation somewhat without losing functionality. I couldn't find a canonical list of installed packages, just additional packages to install on top of another base (Raspbian by any chance?).

kingosticks, 2015-03-06:
You'll have to forgive my poor attempt at humour there. We are going to move to a packaged version of musicbox in the very near future which will make the update process much easier (although you may run into problems when running apt-get with that much free space). My only concern is continually having to worry about making everything fit in the 1 GB case when adding new packages etc.  

But the idea of an image with a read-only partition for the majority, alongside an expandable writeable area for user data has come up before. I think that's a nice idea. Although does it still let people customise the install and install some other packages if they choose? How does that work?

There is a script in the root of the repository which should show the steps to create an image (yes, from base Raspbian). Any review of stuff to trim would certainly be helpful. That's all currently done by @woutervanwijk so he probably has a better handle on how feasible it all is.

hadess, 2015-03-06:
@kingosticks any data space will eventually be filled up ;)

The default Raspbian image is probably huge compared to what needs to be on the default MusicBox system. I mean, there's probably all sorts of UI libraries and applications on the default install which MusicBox doesn't need.

(I'm used to working with Fedora's Live images, and in 1.4Go, we fit a full graphical system, Firefox and LibreOffice)

hadess, 2015-03-08:
What's the canonical way of creating Pi MusicBox? create_musicbox.sh or "Create Pi MusicBox.rst"? (or are they the exact same). I'm half-way through preparing a scripted way to create an image, rather than using a default Raspbian image as a base.

kingosticks, 2015-03-08:
I think they are the same.

woutervanwijk, 2015-03-09:
"Create Pi MusicBox.rst" is a document describing how to create the image, the shell script is a way of automating that. They could certainly be merged.
MusicBox uses a stripped version of Raspbian. The script removes loads of packages that are not needed. The next version could be a bit smaller, that's easy. I'll try that.

hadess, 2015-03-10:
Using the network installer I mentioned, I got to an installed size of 567 megs for the root, and 50 megs for /boot (28 megs actually used).

hadess, 2015-03-11:
It boots (in QEmu)!
https://github.com/hadess/raspbian-ua-netinst

A number of items still to fix:
- Fix overriding RPi version to target:
https://github.com/debian-pi/raspbian-ua-netinst/issues/194
- Fix filesystem detection(s):
https://github.com/debian-pi/raspbian-ua-netinst/issues/193#issuecomment-77955462
- Find a newer QEmu kernel that does have modules available for it
- Find why the image takes 1.2GB on the filesystem (for du), when there's only 600 megs used.

hadess, 2015-03-11:
You'll probably want to modify the proxy in installer-config.txt (or better, install apt-cacher-ng so you don't wait ages for the 200 megs of packages to be installed).

hadess, 2015-09-10:
By the way http://www.woutervanwijk.nl/pimusicbox/ says:
`The image is for a 1GB card`
Would be good to change it until this bug is fixed (recommend 2GB?)

jodal, 2015-09-13:
http://www.woutervanwijk.nl/pimusicbox/ is an old outdated version of the site, and not the same as http://www.pimusicbox.com/.

@woutervanwijk Can you replace http://www.woutervanwijk.nl/pimusicbox/ with a redirect to http://www.pimusicbox.com/ ?

woutervanwijk, 2015-09-14:
Sure!

Verstuurd vanaf mijn iPad

woutervanwijk, 2015-09-14:
BTW, I chose to support 1G cards because quite some people asked about it (they had a 1G card lying around). The 0.6 image, I think has some 110+ mb free space, so it could be made a bit smaller. Maybe the best way is to create an image that is as small as possible and let it resize to the size of the card at first boot (like it does now, but maybe implemented a bit better)

kingosticks, 2015-09-14:
Removing the wolfram stuff is the lowest hanging fruit here, it is huge and
completely unnecessary for us. In a related note, I think that expanding
the filesystem on first boot needs to be automatic (along with some other
stuff) rather than an option in settings.ini. If we want to disable it
during testing then we  can find a way to do that.

jcass77, 2016-02-22:
I don't think this is the direction that we are heading in for Musicbox 1.0: we won't be distributing our own full image but relying on the package management system. 

Even if we base the distribution on [Raspbian Jessie Lite](https://www.raspberrypi.org/downloads/raspbian/), I think the installed size will still be at least 1.3Gb?

Am I right @kingosticks?

jcass77, 2016-02-22:
Also see #324.

closed 2016-02-22 by jcass77