pimusicbox/pimusicbox #238: Chromecast support

kingosticks, 2015-03-02:
Some sort of chromecast support might be useful. There a few ways this might work:
1. From another device to musicbox
2. From musicbox to a chromecast dongle (presenting it as another soundcard). 
3. Combination of the above. 

Personally I think the former is more useful and would complement the airplay support nicely. 

I'm unsure how feasible any of this is so for now this is more of a placeholder for collecting relevant ideas. Or at least documenting why we don't have this feature. 

I think option 2 is what's discussed at https://github.com/abarisain/dmix/issues/697

jkaberg, 2015-03-26:
More fuel for the fire ;-)

https://github.com/dz0ny/leapcast

hadess, 2015-03-26:
leapcast is awful, it requires to run Chrome (possibly Chromium) on the machine.

kingosticks, 2015-09-30:
If we used pulseaudio then https://github.com/masmu/pulseaudio-dlna looks nice to send audio to a chromecast (and also dlna devices).

hadess, 2015-09-30:
> If we used pulseaudio

We don't already? I thought we did. If not, could you create an issue about using it?

kingosticks, 2015-09-30:
No we do not, wheezy has pulseaudio v2 which is apparently too old to be worth the hassle. But jessie has v5. We'd just need to investigate any performance impact of using it though, I've no idea where it stands in regards to that, particularly for the older models.

kingosticks, 2016-02-05:
https://github.com/xat/castnow might be interesting too.

kingosticks, 2016-10-06:
mkchromecast

bastian94, 2017-02-12:
Any updates about this?

kingosticks, 2017-02-12:
There is still nothing I know of that will allow pimusicbox to act as a Chromecast receiver (option 1), we can thank Google for that. 

Option 2 I'm less interested in right now as streaming *from* pimusicbox to other devices is not something we currently support by any means. I'm not saying we won't support that functionality in the future, but 'casting' is just one of many different technologies we could support and it's not necessary the most worthwhile or the best one to start with. But that all depends on what the options are so if someone was to look into it then we could push the idea along.

bastian94, 2017-02-12:
Ok

kingosticks, 2017-05-28:
Mkchromecast seems to support casting without pulseaudio https://github.com/muammar/mkchromecast/blob/master/README.md

hoveeman, 2018-03-12:
I'm going to try this with mkchromecast. Was anyone successful getting this to work? I basically want to route all audio from the pi to a chromecast audio group.

BLX-S, 2018-03-22:
I would add that I find option 1 to be much more valuable. I really like the seemless chromecast audio integration that you can get, and if only we could integrate that into PMB...

kingosticks, 2018-03-22:
I agree but I still don't know of a way to achieve it.