pimusicbox/pimusicbox #240: static_dir configuration parameter overridden by other extensions

jcass77, 2015-03-03:
Using the ```static_dir``` configuration parameter, as is currently being done to specify the default Mopidy webclient, does not work as intended for every scenario. See https://github.com/mopidy/mopidy/issues/860 for details.

The problem is that all extensions are served from the mopidy root folder, which means that any static content folder that shares a name with whatever extensions are installed will not be served as expected. An example of this is the https://github.com/tkem/mopidy-local-images extension which is named 'images' and clashes with the 'images' folder of the webclient.

To test this try accessing the MusicBox web client in the following ways after installing mopidy-local-images:

1. http://musicbox.local -> ```static_dir``` reference is used, images folder is overridden by mopidy-local-images, and the relative URL to the default cover art in the footer (images/default_cover.png) cannot be found

2. http://musicbox.local/musicbox_webclient -> standard mopidy extension configuration is used, and images are served correctly from /musicbox_webclient/images/default_cover.png

Until there is a reliable way of specifying defaults in mopidy, It is probably best to move away from using ```static_dir``` and just revert to using the URL in (2) above?

Note that this issue should only become evident should we decided to merge woutervanwijk/Pi-MusicBox#234 and make mopidy-local-images part of the standard Pi MusicBox distribution.

kingosticks, 2015-03-03:
I think we currently do port redirection with IP tables but this is no good if we want URL redirection too. Is this the job for a reverse proxy? I've no experience with any of this but google shows me
* Tinyproxy
* Nginx
* Apache with mod_proxy

I suppose we just want something small and really simple so probably not Apache. I wonder if it would break any clients. Presumably musicbox-webclient is now OK thanks to https://github.com/woutervanwijk/Mopidy-MusicBox-Webclient/pull/66

Maybe this would also easily allow us to password protect certain pages (i.e. the settings and system pages). Although I only know of [one request for this feature](https://discuss.mopidy.com/t/use-pi-musicbox-for-streming-a-live-radio-broadcast-stream-in-different-rooms-in-a-school/583).

jodal, 2015-03-03:
+1 to nginx

It also supports proxying WebSockets, which is quite important in this case.

kingosticks, 2015-03-06:
I played around with this today and struggled a bit, ended up needing a rewrite section to make it work and I can't help but thinking that's overkill:
```
server {
    listen 80;
    server_name $hostname;

    location / {
      proxy_redirect          off;
      proxy_set_header        Host $host;
      proxy_set_header        X-Real-IP $remote_addr;
      proxy_set_header        X-Forwarded-For $proxy_add_x_forwarded_for;
      proxy_set_header        X-Forwarded-Proto $scheme;
      rewrite                 /(.*) /$1 break;
      proxy_pass              http://localhost:6680/musicbox_webclient/;
    }
}
```

So if we want to make it nice and simple to change the default client (like it currently is) we'll need something in startup.sh to hack the nginx config file. I'm also  very unimpressed you can't have macros in your nginx config files, that would have made the hack a bit simpler.

kingosticks, 2015-03-06:
turns out that only worked for musicbox-webclient. Changing the proxy_pass to mopify and moped made it redirect to the /mopidy/ web client list. I'm rapidly losing interest in this.

jcass77, 2015-03-07:
I got more or less the same results trying to use ```proxy_pass```, with many of the moped elements not rendering as expected.

Perhaps we can consider doing a simple URL redirect as per the below:

```
server {
    listen       80;
    server_name  localhost;
    return       301 http://$host:6680/musicbox_webclient/;
}
```

This will just redirect the browser to the web client as if the above link had been clicked on the mopidy landing page. It is not as elegant as using ```proxy_pass``` unfortunately...

jcass77, 2015-03-07:
On the ```proxy_pass``` issues: I think the problem is websocket related. My browser's error console shows:

WebSocket connection to 'ws://192.168.0.105/mopidy/ws' failed: Error during WebSocket handshake: 'Connection' header value is not 'Upgrade'
moped-0.4.2.js:9:40589

I tried adding this explicitly by including the following in the nginx configuration file:
```
proxy_http_version 1.1;
proxy_set_header Upgrade $http_upgrade;
proxy_set_header Connection "upgrade";
```
But it appears that this is only supported from [version 1.3.13](http://nginx.org/en/docs/http/websocket.html) onwards.

kingosticks, 2015-03-07:
@adamcik showed me some nginx magic along those lines on IRC.
http://dpaste.com/2B0JQRG

On the proxy_pass issues: I think the problem is websocket related. My
browser's error console shows:

WebSocket connection to 'ws://192.168.0.105/mopidy/ws' failed: Error during
WebSocket handshake: 'Connection' header value is not 'Upgrade'
moped-0.4.2.js:9:40589

I tried adding this explicitly by including the following in the nginx
configuration file:

proxy_http_version 1.1;
proxy_set_header Upgrade $http_upgrade;
proxy_set_header Connection "upgrade";

But it appears that this is only supported from version 1.3.13
<http://nginx.org/en/docs/http/websocket.html> onwards.

adamcik, 2015-03-07:
Oh yeah, I forgot the mention that I've been using the nginx PPA so I ha the latest stable, instead of whatever Ubuntu LTS has.

jcass77, 2015-03-08:
It doesn't appear as if there are any updated nginx packages available for the Raspberry Pi. I managed to build version 1.6.2 from source using [these instructions](http://elinux.org/RPi_Nginx_Webserver) (just omit --without-http_proxy_module when doing ```./configure```)

jodal, 2015-03-08:
http://archive.raspbian.org/raspbian/pool/main/n/nginx/ has both 1.2.1 (wheezy) and 1.6.2 (jessie). You can probably download it by hand and install the deb file with `dpkg -i path/to/deb`.

jcass77, 2015-03-08:
I think these nginx server settings can be used to map traffic on port 80 to 6680, and redirect the user to the default web client. Please check and provide inputs?

```
map $http_upgrade $connection_upgrade {
    default upgrade;
    '' close;
}

server {
        #listen   80; ## listen for ipv4; this line is default and implied
        #listen   [::]:80 default_server ipv6only=on; ## listen for ipv6

        # Make site accessible from http://localhost/
        server_name localhost;

        # Enable websocket proxying for mopidy
        location /mopidy/ws {
                proxy_pass http://localhost:6680;
                proxy_http_version 1.1;
                proxy_set_header Upgrade $http_upgrade;
                proxy_set_header Connection "upgrade";
                proxy_read_timeout 604800;
                proxy_send_timeout 604800;
        }

        # Redirect root folder to the Mopidy MusicBox Webclient
        location = / {
                rewrite ^ http://$host/musicbox_webclient/;
                proxy_pass http://localhost:6680/;
                proxy_http_version 1.1;
        }

        # Redirect all requests on port 80 to port 6680
        location / {
                proxy_pass http://localhost:6680/;
                proxy_http_version 1.1;
        } 
```

jcass77, 2015-03-08:
I've started pull request https://github.com/woutervanwijk/Pi-MusicBox/pull/244 on the 'develop' branch in case we want to start making the changes official.

adamcik, 2015-03-08:
Thanks, that is also more or less what I suggested om IRC. But as Nick already mentioned it needs at least nginx 1.3.13 - so if we get that bit sorted the proposed config should work.

kingosticks, 2015-04-07:
On IRC, @adamcik has helpfully suggested [haproxy as an alternative to nginx](http://blog.silverbucket.net/post/31927044856/3-ways-to-configure-haproxy-for-websockets). Like nginx, it also  [supports authentication](http://serverfault.com/questions/239749/possible-to-add-basic-http-access-authentication-via-haproxy). Unlike nginx, it has websocket support out of the box and doesn't require messing around with jessie packages.

jodal, 2015-04-08:
It should be noted that the target release date for Debian jessie as the new stable is April 25, so nginx with websocket support out of the box will soon be the default.

DavisNT, 2015-04-09:
IMHO if the only issue is default web GUI, then most simple solution would be setting *static_dir* in start-up script to a folder that makes HTML meta redirection to required GUI.
I would say that URL rewriting in reverse proxy is something that might cause trouble at some later point (e.g. with some naughty JavaScript) and I **suggest avoiding it** in the design. I would like to suggest HTTP redirection instead.
If there is a plan adding authentication, then reverse proxy sounds like a good idea. I personally like HAProxy more (due to security considerations).

kingosticks, 2015-04-09:
The musicbox Web client (along with other clients people may wish to use
instead) is now a Web app rather than static content. So that, along with
the original problem in this issue, means static_dir isn't really an option
anymore.

DavisNT, 2015-04-10:
@kingosticks I know that MusicBox Webclient does  not work as static page anymore. My suggestion about *static_dir* and HTML meta redirect is working (tested) and documented in section "4. Fix-ups after Python packages update" of https://github.com/woutervanwijk/Pi-MusicBox/wiki/Updating-components-to-latest-versions

kingosticks, 2015-04-10:
While I appreciate the effort, we currently don't advise any upgrading and
I don't have time to support people with it. Does your command prevent
updating Mopidy to v1.0? If not, it'll break people using mopidy-subsonic
and quite possibly other things.

DavisNT, 2015-04-10:
@kingosticks Thanks for info! I have added warnings to Wiki article. My command does **not** prevent updating to Mopidy 1.0 - the command updates all Python packages to latest versions available in PyPI.

Back to topic of redirection, these commands (taked from Wiki article) create HTML meta redirect (regardless of Mopidy and Mopidy-MusicBox-Weblcient versions):

    sed -i 's/static_dir \=.*/static_dir = \/opt\/webclient/g' /boot/config/settings.ini
    rm /opt/webclient
    mkdir /opt/webclient
    echo -e "<html><head><meta http-equiv=\"refresh\" content=\"0; URL=/musicbox_webclient/index.html\"></head><body>Web interface moved, <a href=\"/musicbox_webclient/index.html\">click here</a></body></html>" > /opt/webclient/index.html

When user opens http://musicbox.local/ he is automatically redirected to http://musicbox.local/musicbox_webclient/index.html

kingosticks, 2015-04-11:
It is certainly much simpler. Any reason not to go with @DavisNT's suggestion? We can then consider authentication as a separate issue.

jcass77, 2015-04-19:
I think this is a good workaround while we wait for proper nginx packages to become available for the Raspberry Pi.

Perhaps we can keep the pull request with the nginx configuration changes in case we want to add it later (along with the user authentication settings)?

jcass77, 2016-02-22:
Closing this as the redirect workaround seems to be working well enough for everyone at the moment.

closed 2016-02-22 by jcass77