pimusicbox/pimusicbox #241: MusicBox not visible as an available rendered via UPNP

jhenkins, 2015-03-06:
Hello,

A bit of a frustration:

Ever since I installed 0.5.1 (also 0.5.3 and 0.5.4b), I cannot use MusicBox as an audio renderer via UPnP, using the following UPnP utilities:

1. Leia (Windows-based UPnP controller, works under Wine too)
2. UPnP Tool - a simple UPnP debug tool for Android
3. BubbleUPnP player for Android

I use MiniDLNA on Linux as my main media library.

Any idea what could be the issue? Please let me know what debug information I can add to help RCA.

Thank you in advance,
Jan Henkins

kingosticks, 2015-03-28:
i think we'll make more progress on this by posting on mopidy.discuss.com as i know the upmpdcli dev has been active on there.

jhenkins, 2015-05-29:
Thank you for the comment. I will reconfirm the issue, and start a discussion there if the problem persists.

closed 2015-05-29 by jhenkins