pimusicbox/pimusicbox #242: Default to C.UTF-8 locale

hadess, 2015-03-08:
Seems that the current default is en_US.UTF-8 which isn't installed by default. C.UTF-8 is however, and is a much better fit for MusicBox, with all the non-ASCII band names :)
```
$ cat /etc/default/locale 
#  File generated by update-locale
LANG=en_US.UTF-8
```

kingosticks, 2015-03-09:
I wasn't aware of much difference between them. As I understand it, its just how upper and lower case non ASCII chars are sorted.and some uninteresting currency stuff. Whatever is easier I guess.

hadess, 2015-03-09:
Between which? The main difference between C.UTF-8 and en_US.UTF-8 is that the image has support for the former, and no support for the latter.

kingosticks, 2015-03-09:
I was addressing the comment about band names, and only out of curiosity
since being unsupported is a pretty big issue. I've not generated any
images before so I don't know of what happened there previously.

hadess, 2015-03-09:
I'll make a note of fixing this with my image creation tool.

hadess, 2016-05-24:
```
# dpkg-reconfigure locales
/usr/sbin/dpkg-reconfigure: locales is broken or not fully installed
# locale-gen "en_US.UTF-8"
-bash: locale-gen: command not found
```

Work-around is:
```
# apt-get update ; apt-get install locales
```

Which should install the locales, and build the en_US.UTF-8 locale. If it's not listed in the output, run:
```
dpkg-reconfigure locales
```