pimusicbox/pimusicbox #243: Don't wait for the network to start mopidy

hadess, 2015-03-08:
In fact, we should start mopidy ASAP, as it's likely to be what handles the GPIO connected buttons, LED screens, and various other input/outputs that the device could be connected to.

jodal, 2015-03-08:
Lots of Mopidy's backends does network stuff on startup, and doesn't handle not having network very well.

-1

kingosticks, 2015-03-08:
I'm with jodal on this.

hadess, 2015-03-08:
With NetworkManager and a new enough GLib you can be told when the network comes on, and when it's actually connected to the Internet.

With GLib 2.32, you can tell whether you can connect to a specific server, and you can tell when the network has been turned on or off through the "network-available" property and can_reach() functions. This doesn't rely on NetworkManager.

With GLib 2.44 (coming out as 2.44 in less than a month), you can be told when you're actually connected to the internet:
https://developer.gnome.org/gio/2.43/GNetworkMonitor.html
See the "connectivity" property. This uses NetworkManager.

hadess, 2015-03-08:
I should have mentioned that I think that the backends should gain network awareness so that those that don't can work. In grilo, I transparently hid sources from view if the network was available and they advertised that they needed the network.

jodal, 2015-03-08:
I don't disagree in principle. It's just that this requires quite a bit of work. Given the current state of things, starting Mopidy before the network is up is not recommended.

hadess, 2015-03-08:
Do you have any examples of affected backends that are shipped by default in Pi-MusicBox?

jodal, 2015-03-08:
I know Mopidy-Scrobbler fails permanently to connect to Last.fm if it can't lookup in DNS when starting.

Mopidy-Spotify handles network dropouts nicely once running, but I'm not sure how it will work if it doesn't have network the first time it tries to login.

There may be more issues, but that's the two I know of straight away.

kingosticks, 2015-03-19:
I don't think there is anything we can do about this here.

closed 2015-03-19 by kingosticks