pimusicbox/pimusicbox #248: Very high volume in Mopidy

kingosticks, 2015-03-13:
Taken from https://github.com/woutervanwijk/Pi-MusicBox/issues/216#issuecomment-78213008

> There is no problem if I use AirPlay, but via the webinterface 3/100 is loud as 50% via AirPlay.
>
>Yesterday I found a solution.
>I changed in /boot/config/settings.ini from "mixer = software" to "mixer = alsamixer" and
>
>[alsamixer]
>card = 
>control =
>
>to
>
>[alsamixer]
>card = 1
>control = Master

kingosticks, 2015-03-13:
I think setsound sets the hardware volume to 96%. The default software mixer is set to 85% and this should just reduce Mopidy's volume going into the hardware. Switching to alsamixer will not do any software mixing and so the volume across the whole system should be the same. I'm not sure sure why we don't just default to use the alsamixer (and make setsound adjust the alsamixer config accordingly).

But that said, I would have though the sofware mixer volume should have reduced the volume, not increased it....

artistro08, 2015-03-27:
I also can confirm this issue too. If we can't get mopidy's volume at a decent level, could we just get shairport's output to a tad louder, that would be awesome.

kingosticks, 2017-10-17:
Related to this with a possible solution is https://discuss.mopidy.com/t/pi-musicbox-volume-control-for-amp-hat-not-working-with-spotify-connect/1840