pimusicbox/pimusicbox #251: USB audio broken on 0.5.4beta

coreyk, 2015-03-20:
Fresh install of 0.5.4beta to previously working RPi 1 B with Audioquest Dragonfly DAC
Automatic audio selection does not output to USB out. 
Selecting USB audio and rebooting causes GUI to not load and http://musicbox.local to be unresponsive, though I can still ssh into the Pi.
lsusb shows Dragonfly

kingosticks, 2015-03-20:
Please replace setsound with the fixed version. Details at https://github.com/woutervanwijk/Pi-MusicBox/issues/216#issuecomment-75832516

If it's still an issue then post the output of 'aplay -l'.

nilicule, 2015-03-20:
Cheers, was having the same problem with my USB audio, though manually setting USB didn't cause my PMB to become unresponsive.

Will give this a try and report back.

kingosticks, 2015-03-20:
I'm a bit puzzled why it would become unresponsive, it should have just
used analog and got on with things.

nilicule, 2015-03-20:
The proposed fix indeed fixes the problem, @kingosticks

coreyk, 2015-03-21:
Yes, this fixes the issue of the USB sound card not working.
The webclient not loading must have been due to the apt-get update I tried before starting fresh.

kingosticks, 2015-03-22:
If you want to use the alsamixer you need to configure it for your particular device. Nothing we can do here.

closed 2015-03-22 by kingosticks