pimusicbox/pimusicbox #258: Every instance of track dissapears from queue with "remove from queue".

totalrisp, 2015-03-27:
When I open a playlist and add a song from within that playlist to the queue, this song appears there. If I repeat this, I can have the song appear twice in a row on the queue, and if I do it again, it will be there three times. Works like I think it should. However, if I now decide that I only want two istances of the song in the queue, and go to remove one isntance, PiMusicBox removes every single isntance of that song in the queue. 

It is probably rare for people to add the same song many times in a row like this, but still I don't know if this is intentional or some kind of bug?

kingosticks, 2015-03-27:
It's a bug, its fixed in a PR but sadly it didn't make it in time for the
latest release. There will be another release very soon.

kingosticks, 2015-03-27:
https://github.com/woutervanwijk/Mopidy-MusicBox-Webclient/pull/88

closed 2015-03-27 by kingosticks