pimusicbox/pimusicbox #259: Problem playlists and non-English characters

totalrisp, 2015-03-27:
There are some song titles (and file names) in the music directory which contain umlauted symbols (such as ä,ö, ü). These are displayed correctly and play if they are browsed to from "browse" or from "search". However, when they are added to a playlist they are not displayed correctly and not played. Other songs from the same directory are, but not the ones whose file names contain umlauts.

kingosticks, 2015-03-27:
Just to clarify by 'added to a playlist'  you mean added to the play queue
(aka tracklist)?

totalrisp, 2015-03-27:
No. Sorry, I was not clear enough. I mean in a m3u-file in the playlists directory.

totalrisp, 2015-03-31:
I think the reason is that the locale is GB.utf8, which does not lead to any problems on the sqlite, but since playlists are just lists of files, then this some how comes into effect.

jodal, 2015-08-31:
The M3U support in Mopidy has seen lots of changes since Mopidy 0.19.5, so this will probably work better/differently when Pi MusicBox upgrades to Mopidy 1.x.

xxx1983, 2016-08-30:
I experience the same problem. Because my spotify playlists often suddenly disappear I want to make local playlists in Mopidy. I try to do this using Mopidy Mobile.

In the logfile and on my phone I get the following error:
UnicodeDecodeError: 'ascii' codec can't decode byte 0xed in position 4: ordinal not in range(128)

Is ther anything I can do at this moment to solve it?