pimusicbox/pimusicbox #261: Update Mopidy-Scrobbler

liftarn, 2015-03-30:
The Mopidy-Scrobbler used is 1.1.0, but there is a big fix in 1.1.1 so perhaps it should be updated.

kingosticks, 2015-03-31:
Is this true as of 0.6RC?

liftarn, 2015-04-01:
I haven't checked it, but it appears so.

kingosticks, 2015-04-01:
I think we just normally pull in the latest versions of everything but
since Musicbox v0.6 uses Mopidy v0.19.5 rather than the new v1.0 release,
it was more complicated this time. Either way, next Musicbox release should
see the latest versions included.

jodal, 2015-04-02:
Mopidy-Scrobbler is a frontend, and does as such not have a Mopidy-1.0-release.

Mopidy-Scrobbler 1.1.1 is from December and fixes a compatibility issue with 0.19.5. The full changelog is:

> Updated to work with `None` as the default value of `track_no` in Mopidy's `Track` model. This was changed in Mopidy 0.19.5. (Fixes: #7)

In retrospect, it was of course wrong to change that in Mopidy 0.19.5. It should have waited for the next feature release.

kingosticks, 2015-04-02:
Of course and it's completely fine to manually update with `pip install --upgrade mopidy-scrobbler`.  I just meant that due to the extra complications in the Musicbox 0.6 release we might have mistakenly pulled in an older release of mopidy-scrobbler.

jcass77, 2018-04-11:
Pretty sure that we are using Mopidy-Scrobbler v1.1.1 in the latest 7.0 release candidate.

@kingosticks, note that [v1.2.0](https://github.com/mopidy/mopidy-scrobbler#v120-2018-04-01) was recently released as well in case another update is due.

closed 2018-04-11 by jcass77

kingosticks, 2018-04-11:
Excellent, thanks for the pointer. (https://github.com/pimusicbox/pimusicbox/commit/987ddcf5d7f8e34389b1f93482c10c8d9fe0a4e6)