pimusicbox/pimusicbox #264: Browse does not find local music that is found by other means

totalrisp, 2015-04-03:
Using the following from webclient: browse, local, folder, shows nothing at all in anywhere, while the folders are there and can be reached via other browse options (album, artist etc.)

from mopidy.log:
2015-04-03 18:39:30,913 - ERROR    Error browsing local:directory:Network: 'ascii' codec can't decode byte 0xc3 in position 2: ordinal not in range(128)

kingosticks, 2015-04-03:
Could you try updating sqlite? Run 'pip install
Mopidy-Local-SQLite==0.9.3'.

totalrisp, 2015-04-03:
Ok thanks, issue resolved. I didn't even consider the possibility that 0.6RC does not come with the latest SQLite extension.

kingosticks, 2015-04-03:
Yeh, it's unfortunate but it does not. Thanks for letting us know it's
fixed.

closed 2015-04-15 by kingosticks