pimusicbox/pimusicbox #265: Bluetooth isn’t start

Dvsat, 2015-04-04:
@WoutervanWijk, earlier you said as an answer for connecting a (USB) Bluetooth for the RaspberrPi/Musicbox ; It could work if you follow the instructions. Use /etc/asound.conf instead of ./asoundrc. 
Thats we understand, right and Okay. But then you said; You probably have to disable audio autodetection in /opt/musicbox/startup.sh because it overwrites /etc/asound.conf.
 Again you have right that it will be overwrites, but could you give a howto ? Two questions about this:

1) Could you show or detail the line or command you mean to disable autodetection into startup.sh that will course overwriting ?
2) Wich MusicBox-settings we need more to give the bluetooth dongle the right input and start (mopidy_btmanager ??) 

Thanks in advance,

PS: After the command : > sudo /etc/init.d/bluetooth restart
Answer will be: 
[ ok ] Stopping bluetooth: rfcomm /usr/sbin/bluetoothd.
[ ok ] Starting bluetooth: bluetoothd rfcomm.

Dvsat, 2015-04-05:
Maybe someone can say more about sending the MP3 stream audio input into the bluetooth-dongle ??

There will be connection between the RPi bluetooth-dongle and the bluetooth speakers, look; 
root@MusicBox:~# bt-audio -c 48:5A:B6:A8:1C:A2
Connecting to an audio service
Audio service is connected  (bluetooth led Okay)

The musicbox plays all sounds by Analog and HDMI except the Bluetooth-dongle.
Is there a HowTo to connect the audio into the Bluetooth-dongle also ?

kingosticks, 2015-04-15:
If you have not already found it, writing `/etc/asound.conf` is done [here](https://github.com/woutervanwijk/Pi-MusicBox/blob/4d3d304c8c459a130df53baf29a938ea17ea2e56/filechanges/opt/musicbox/setsound.sh#L200-209) and autodetection in general is called from [here](https://github.com/woutervanwijk/Pi-MusicBox/blob/4d3d304c8c459a130df53baf29a938ea17ea2e56/filechanges/opt/musicbox/startup.sh#L113).

I've never tried to use a bluetooth speaker but hopefully you by now looked at https://discuss.mopidy.com/t/bluetooth-streaming/546/5 for help.

closed 2015-04-15 by kingosticks