pimusicbox/pimusicbox #270: Add Mopidy-AlarmClock package

DavisNT, 2015-04-11:
Package [Mopidy-AlarmClock](https://pypi.python.org/pypi/Mopidy-AlarmClock/) provides alarm clock. :bowtie: :bell: 

Currently the package can be manually set up as described in https://github.com/woutervanwijk/Pi-MusicBox/wiki/Setting-up-Alarm-Clock

Integration requires:
* Adding package [Mopidy-AlarmClock](https://pypi.python.org/pypi/Mopidy-AlarmClock/)
* Adding ability to set timezone via web interface https://github.com/woutervanwijk/mopidy-websettings/issues/8 
* Working NTP package for improved clock synchronization (see https://github.com/woutervanwijk/Pi-MusicBox/wiki/Setting-up-Alarm-Clock for Monit config file)

DavisNT, 2015-04-23:
Is alarm clock feature welcome in Pi MusicBox?

kingosticks, 2015-04-23:
I'm really unfamiliar with the particular extension other than what has already been highlighted at https://github.com/mopidy/mopidy/pull/1085#issuecomment-90399101. Don't think we've had any requests for the functionality either so it's not hugely interesting.

DavisNT, 2015-04-23:
Users interested in this feature, please give your feedback here!

When this feature will be welcome, I will possibly implement it. Meanwhile everybody can manually install alarm clock as described in: https://github.com/woutervanwijk/Pi-MusicBox/wiki/Setting-up-Alarm-Clock

michaellunzer, 2015-05-20:
@DavisNT I'm very interested! I was able to add your solution as you suggested however it would be nice to see the alarm section completely integrated into @woutervanwijk's pimusicbox frontend (maybe as a new tab
:+1: :+1: :+1: :+1: :+1: :+1: :+1: :+1:

DavisNT, 2015-05-24:
@kingosticks Is a Pull Request for MusicBox Webclient addling a link to AlarmClock (if it is installed) in main menu welcome?

kingosticks, 2015-05-26:
I guess it's ok to include a navigation link to it which we'll show only if it's installed (don't want to create another config option for this).

michaellunzer, 2015-05-28:
I can't wait! Let me know if I can help in any way @DavisNT

DavisNT, 2015-05-30:
I have created a Pull Request that adds link to Alarm Clock in Musicbox Webclient. 
@michaellunzer thanks for Font Awesome Icon!
@kingosticks could you please do the review and merge?

jcass77, 2016-02-22:
#104 has been merged which addresses this issue.

closed 2016-02-22 by jcass77