pimusicbox/pimusicbox #273: Increase log verbosity to at least info

adamcik, 2015-04-14:
I know this has been tuned down to save the SD card. But we could easily have a more verbose logging to a small tmpfs. Using https://docs.python.org/2/library/logging.handlers.html#rotatingfilehandler there would also be a upper limit on the size and number of files. This could be instead of or in addition to logging on the SD.

kingosticks, 2015-05-04:
https://docs.python.org/2/library/logging.handlers.html#memoryhandler also sounds pretty interesting.

adamcik, 2015-05-05:
In order to batch up the writes I presume, that should help avoid some of the wear for the memory cells I would hope. This could also be combined with the rotating handler.

kingosticks, 2015-05-05:
Indeed. Am I right in saying that I'd need to set it up programmatically within some extension (by grabbing the root logger) rather than being able to do it through Mopidy's existing logging.conf? 

Thinking again, I don't see how the memory handler would allow us to also provide a debug endpoint that includes the log. So maybe it's a step too far and the tmpfs is the way to go. There is always the posability of a cron job that syncs the tmpfs to the SD card if persistence is deemed important.

kingosticks, 2015-05-05:
Seems actually I don't understand logging.conf one bit but it looks like it can in fact do anything. 

And if I subclass memory handler (or do something like you suggest in https://github.com/woutervanwijk/Pi-MusicBox/issues/274#issuecomment-98974838) then I can do whatever I want with the buffer and make it accessible to the debug/log endpoint.

adamcik, 2015-05-05:
Ah, you are thinking about #274 - In my head these are just to different issues that both would be nice to have. For this case some config in something like logging.conf makes sense. For the other one you probably want code.

kingosticks, 2015-05-05:
Yeh, they are separate I just didn't want to make #274 more difficult to implement with a particular implementation choice here. But it's clear now that there's no problem there.