pimusicbox/pimusicbox #274: Log to memory?

adamcik, 2015-04-14:
Having a `musicbox.local/debug` endpoint might be interesting. This could have `/debug/logs`, `/debug/config` and `/debug/deps` type endpoints that we can point people to for getting debug info without forcing them to go to the CLI.

kingosticks, 2015-04-14:
:+1: to all those.

The logs one was something I was thinking about, I used to really like the way you could see the live log in sabnzbd etc.  But I wasn't sure how to best go about it. Would you have some kind of logging handler defined in the websettngs that returns the last n lines? Or something outside of the mopidy process entirely (and served by something else)? How does logging to memory come into this?

adamcik, 2015-04-14:
That would be one way of implementing it. We could also just have something that serves the log files. Updated the title to reflect what I was suggesting this solve, not how.

adamcik, 2015-05-05:
```
class CircularInMemoryHandler(logging.Handler):
    def __init__(self, capacity):
        super(CircularBufferHandler, self).__init__()
        self.buffer = collections.deque(maxlen=capacity)

    def emit(self, record):
        self.buffer.append(record)
```

is more or less what I was thinking.

kingosticks, 2015-05-07:
Thinking about this today, it would be really nice if we could still serve the log file to the user if Mopidy itself is unable to start. 

If we go with nginx (https://github.com/woutervanwijk/Pi-MusicBox/pull/244) for reverse proxying then it could also serve the log. Nginx doesn't support CGI itself but something else could run `mopidy deps|config` and save the output for nginix to serve.