pimusicbox/pimusicbox #275: Mopify Web Client

hkultraman, 2015-04-15:
Hi,

I would like to use Mopify as web client. But can't find any instruction on it. I am now using stable version of Pi MusicBox 1.0

Can anyone point me to where can i find the instruction to use Mopify on Pi MusicBox Stable v1.0

Link to Mopify: https://github.com/dirkgroenen/mopidy-mopify

Many thanks.

Cheers
Michael

kingosticks, 2015-04-15:
I think you must mean version 0.6. Mopify is a bit of a special case right now due to the way it's implemented. It should be fixed in the next version but until then you can try the following (untested by me) commands from the wiki:
```
sed -i 's/static_dir \=.*/static_dir = \/opt\/webclient/g' /boot/config/settings.ini
rm /opt/webclient
mkdir /opt/webclient
echo -e "<html><head><meta http-equiv=\"refresh\" content=\"0; URL=/musicbox_webclient/index.html\"></head><body>Web interface moved, <a href=\"/musicbox_webclient/index.html\">click here</a></body></html>" > /opt/webclient/index.html
```

But replace `musicbox_webclient/index.htm` with `mopify/index.htm`, or whatever the correct url is.

hkultraman, 2015-04-15:
yup, i mean 0.6 (i have updated my post).

BTW, from mopify, they states it requires:

- Mopidy 1.0
- Mopidy-Spotify 1.3 (or higher)
- enabled the HTTP Extension.

Are these 3 thing already included / enabled in v0.6?

Many thanks.

Cheers

kingosticks, 2015-04-15:
Ahh, Yeh I forgot that slight issue. We don't support mopidy v1.0 yet. If
you install the latest version of mopify it will pull in mopidy 1.0 and
then some things will not work.

However, I think there is already an older version of mopify installed by
default which should be compatible with our 0.19.5 version of mopidy. Or
you can identify and manually install an older version of mopify that
doesn't require mopidy v1.0. Something like `pip install
mopify=some-old-version-number`.

kingosticks, 2015-04-15:
And i of course meant `pip install mopidy-mopify=some-old-version-number`

hkultraman, 2015-04-15:
So, what is the version of mopidy come with Pi MusicBox 0.6?

Thanks.

kingosticks, 2015-04-15:
0.19.5

hkultraman, 2015-04-15:
And v0.6 doesn't include Mopidy-Spotify, isn't it?

many thanks.

kingosticks, 2015-04-15:
Yes it does. Mopidy-spotify has always been included.

kingosticks, 2015-04-15:
If you have access to your system just run `mopidy deps` to see what's included and what versions.

hkultraman, 2015-04-15:
Many thanks, cheers.

kingosticks, 2015-04-15:
Supporting webapp style webclients is tracked at https://github.com/woutervanwijk/Pi-MusicBox/issues/276

tatoosh, 2015-04-26:
Is Mopidy-Mopify now running?
I tryed myself but it did not work :(

hkultraman, 2015-04-26:
Try http://[your_ip]/mopify

tatoosh, 2015-04-26:
it's not working :(

kingosticks, 2015-04-26:
@tatoosh please can you be clearer about what exactly you have done and what exactly is the error/problem.

tatoosh, 2015-04-28:
@kingosticks:
I wantet to install https://github.com/mopidy/mopidy-spotify. But i think it is not compatible. Will you add this!?

kingosticks, 2015-04-28:
As I mentioned previously in this thread, mopidy-spotify has always been a part of Pi-Musicbox. You do not need install anything extra.

kingosticks, 2015-05-04:
Since we already have https://github.com/woutervanwijk/Pi-MusicBox/issues/276 to address the more general case there doesn't seem to be anything else to do here.

closed 2015-05-04 by kingosticks