pimusicbox/pimusicbox #276: Support 'webapp' type webclients

kingosticks, 2015-04-15:
We currently use `static_dir` to select the default webclient with the advisory that this can be modified to point to the user's preferred webclient. This doesn't work for dynamic 'webapp' type webclients such as mopify and the latest Mopidy-Musicbox-Webclient (v1.0.0).  

Related to https://github.com/woutervanwijk/Pi-MusicBox/issues/240

hkultraman, 2015-04-16:
BTW, any plan to include mopify as alternative client?

kingosticks, 2015-04-16:
It should already be included in the base install. Unless I'm mistaken. But
either way, yes.

hkultraman, 2015-04-18:
BTW, is it possible to make port 6680 point to musicbox.local/mopify?

Currently it is pointing to simple web client.

As I find an Android mopify client which need mopify using 6680.

Many thx.

kingosticks, 2015-04-18:
https://github.com/woutervanwijk/Pi-MusicBox/issues/240#issuecomment-91679160