pimusicbox/pimusicbox #279: Add context menu to browse entries

joh-klein, 2015-04-19:
I always miss the option to add browse entries (from spotify or other sources) to the queue like there is with search results.

kingosticks, 2015-04-20:
This was fixed in the webclient in https://github.com/woutervanwijk/Mopidy-MusicBox-Webclient/pull/82

closed 2015-04-20 by kingosticks

joh-klein, 2015-04-20:
Since I checked yesterday with v 0.6.0. When (roughly) is this fix going to be in the master branch of the PI-MusicBox?

kingosticks, 2015-04-20:
I'm afraid I really don't know what the next Musicbox release will be. But if you want, you could update to the latest development version of the webclient with `pip install -U https://github.com/woutervanwijk/Mopidy-MusicBox-Webclient/archive/develop.zip`. It should be stable.