pimusicbox/pimusicbox #281: Pi-musicbox 0.6 webui browser access wont work please help

borneoboy, 2015-05-09:
My musicbox 0.6 on raspberry pi was working fine with pi-dac+, with both ethernet which works fine and wifi which i predominantly used. Since a browser shutdown....Nothing!!!
Ethernet cable connection and static ip musicbox works fine through browser. Wifi wont work at all no matter what i try. Ive tried different images from 0.5.1 through 0.6.
Wlan usb dongle is recognised, drivers are loaded, ssid and pass in settings.ini set ok.....but dead as a dodo....browser wont connect. Have pinged Pi IP address from PC command.....data is sent and received ok.....anyone please help its driving me mad.

kingosticks, 2015-05-09:
So, if you can ping it can you connect to it via ssh? Can you control it using an mpd client? Can you plug it a hdmi cable and see what the error is during startup.

michaellunzer, 2015-05-26:
My Pi recently experienced this error. Any attempt to connect via musicbox.local/ or the IP address results in a connection refused error, but I can connect via SSH and airplay into the device. 

I was about to reinstall musicbox, but I will see what I can find before wiping it. I will have to plug it into an HDMI cable tonight and see what the error is during the startup and report back here

michaellunzer, 2015-05-28:
@borneoboy I had an error with an extension - [ttsgpio](https://github.com/9and3r/mopidy-ttsgpio/)

I changed in the settings:
```
    debug_gpio_simulate = false
```
when I changed it back to true, it fixed the errors!
Hope that helps!

kingosticks, 2015-05-28:
For the record, that extension is not bundled with musicbox.  If you have any extra extensions or have done any modifications yourself then please do mention them else it's really difficult to help.

closed 2015-08-31 by jodal