pimusicbox/pimusicbox #282: crashing while loading artist with search function

TheBouwman, 2015-05-10:
everything freezes when I'm searching for an artist on spotify. The music wouldn't play and the connection to the webclient gets dropped. I takes 5min before I can reconnect to the client.

kingosticks, 2015-05-10:
Are these searches that have a huge number of results? Do you see any
errors in /var/log/mopidy/mopidy.log?

TheBouwman, 2015-05-10:
The log doesn't show anything. But when you browse the artist you see the albums with the tracks sorted below them. If you open the album first and then play the song everything works fine but when I play a track from the same album on the artist page everything stopped working.

TheBouwman, 2015-05-10:
here is the log from the last couple of minutes.
https://www.dropbox.com/s/v26nqrzkeozy2v3/mopidy.log?dl=0

closed 2019-06-05 by TheBouwman