pimusicbox/pimusicbox #285: Remove monit

kingosticks, 2015-05-28:
Do we still need monit? There are no known bugs causing Mopidy to hang or crash, this extra complexity and load doesn't seem necessary.

jcass77, 2015-06-14:
I've got a bunch of Mopidy restart messages in ```/var/log/monit.log```, but this could just be because I was hacking around (although I always try to stop monit first).

If you are not seeing any restarts being performed by monit then I agree, and we should probably take it out.

mwild1, 2015-07-24:
Hmm, I'm in favour of removing complexity, but I would argue that any decent consumer device would have a watchdog or similar to ensure that the device remained functional in spite of unexpected issues (and let's face it, software always has unexpected issues).

If monit really is causing extra load or complexity, are there alternatives? I've used daemontools in the past, which can automatically restart processes when they exit.

kingosticks, 2015-08-12:
I'm still not sure what we need monit to do but we don't need it to sit there polling the http interface every 5 seconds (or whatever it's currently doing). I would really like to know some usage stats on how often it's stepped in and restarted Mopidy, excluding those reported time it's done so unnecessary and caused people issues.

jcass77, 2015-08-14:
I've included an extract of my monit.log file below. I keep Mopidy running 24/7 on a Pi Musicbox - it looks like monit does a restart about once a week (with a lot of manual restarts in between).

Not sure if it is worth the additional complexity or hassle, but it does seem to be doing its thing every now and then:

[SAST Aug  6 05:43:42] info     : Starting monit daemon
[SAST Aug  6 05:43:42] info     : Monit start delay set -- pause for 300s
[SAST Aug  6 05:49:00] info     : 'system_MusicBox' Monit started
[SAST Aug  6 19:35:17] error    : 'mopidy' failed protocol test [HTTP] at INET[localhost:6680] via TCP -- HTTP: Error receiving data -- Resource temporarily unavailable

[SAST Aug  6 19:17:20] info     : Starting monit daemon
[SAST Aug  6 19:17:20] info     : Monit start delay set -- pause for 300s
[SAST Aug  6 21:59:03] info     : 'system_MusicBox' Monit started
[SAST Aug  7 05:49:10] info     : monit daemon with pid [2700] killed
[SAST Aug  7 05:49:10] info     : 'system_MusicBox' Monit stopped
[SAST Aug  7 05:49:24] info     : Starting monit daemon
[SAST Aug  7 05:49:24] info     : Monit start delay set -- pause for 300s
[SAST Aug  7 05:54:43] info     : 'system_MusicBox' Monit started
[SAST Aug  7 06:26:44] error    : 'mopidy' process PID changed from 2744 to 3141
[SAST Aug  7 06:26:44] error    : 'mopidy' failed, cannot open a connection to INET[localhost:6680] via TCP
[SAST Aug  7 06:28:44] info     : 'mopidy' process PID has not changed since last cycle
[SAST Aug  7 08:32:53] error    : 'mopidy' failed protocol test [HTTP] at INET[localhost:6680] via TCP -- HTTP: Error receiving data -- Resource temporarily unavailable

[SAST Aug  7 08:34:58] error    : 'mopidy' failed protocol test [HTTP] at INET[localhost:6680] via TCP -- HTTP: Error receiving data -- Resource temporarily unavailable

**[SAST Aug  7 08:34:58] info     : 'mopidy' trying to restart**
[SAST Aug  7 08:34:58] info     : 'mopidy' stop: /etc/init.d/mopidy
[SAST Aug  7 08:35:29] info     : 'mopidy' start: /etc/init.d/mopidy
[SAST Aug  7 08:37:29] info     : 'mopidy' connection succeeded to INET[localhost:6680] via TCP
[SAST Aug 10 13:28:47] error    : 'mopidy' failed protocol test [HTTP] at INET[localhost:6680] via TCP -- HTTP: Error receiving data -- Resource temporarily unavailable

**[SAST Aug 10 13:28:47] info     : 'mopidy' trying to restart**
[SAST Aug 10 13:28:47] info     : 'mopidy' stop: /etc/init.d/mopidy
[SAST Aug 10 13:29:17] info     : 'mopidy' start: /etc/init.d/mopidy
[SAST Aug 10 13:31:17] info     : 'mopidy' connection succeeded to INET[localhost:6680] via TCP
[SAST Aug 10 14:28:39] info     : monit daemon with pid [2688] killed
[SAST Aug 10 14:28:39] info     : 'system_MusicBox' Monit stopped
[SAST Aug 10 14:32:13] info     : Starting monit daemon
[SAST Aug 10 14:32:13] info     : Monit start delay set -- pause for 300s
[SAST Aug 10 14:37:13] info     : 'system_MusicBox' Monit started
[SAST Aug 11 19:22:59] info     : monit daemon with pid [5417] killed
[SAST Aug 11 19:22:59] info     : 'system_MusicBox' Monit stopped
[SAST Aug 11 19:23:12] info     : Starting monit daemon
[SAST Aug 11 19:23:12] info     : Monit start delay set -- pause for 300s
[SAST Aug 11 19:28:29] info     : 'system_MusicBox' Monit started
[SAST Aug 13 15:33:17] error    : 'mopidy' failed protocol test [HTTP] at INET[localhost:6680] via TCP -- HTTP: Error receiving data -- Resource temporarily unavailable

[SAST Aug 14 14:17:37] error    : 'mopidy' process PID changed from 2708 to 6515
[SAST Aug 14 14:19:37] info     : 'mopidy' process PID has not changed since last cycle
[SAST Aug 14 15:55:39] error    : 'mopidy' process PID changed from 6515 to 12634
[SAST Aug 14 15:57:39] info     : 'mopidy' process PID has not changed since last cycle
[SAST Aug 14 16:20:18] info     : monit daemon with pid [2685] killed
[SAST Aug 14 16:20:18] info     : 'system_MusicBox' Monit stopped
[SAST Aug 14 16:20:38] info     : Starting monit daemon
[SAST Aug 14 16:20:38] info     : Monit start delay set -- pause for 300s
[SAST Aug 14 16:25:38] info     : 'system_MusicBox' Monit started
[SAST Aug 14 16:27:12] info     : monit daemon with pid [12858] killed
[SAST Aug 14 16:27:12] info     : 'system_MusicBox' Monit stopped
[SAST Aug 14 16:27:30] info     : Starting monit daemon
[SAST Aug 14 16:27:30] info     : Monit start delay set -- pause for 300s
[SAST Aug 14 16:32:30] info     : 'system_MusicBox' Monit started
[SAST Aug 14 17:03:19] info     : monit daemon with pid [13010] killed
[SAST Aug 14 17:03:19] info     : 'system_MusicBox' Monit stopped
[SAST Aug 14 17:03:38] info     : Starting monit daemon
[SAST Aug 14 17:03:38] info     : Monit start delay set -- pause for 300s
[SAST Aug 14 17:08:38] info     : 'system_MusicBox' Monit started
[SAST Aug 14 17:25:53] info     : monit daemon with pid [13199] killed
[SAST Aug 14 17:25:53] info     : 'system_MusicBox' Monit stopped
[SAST Aug 14 17:26:09] info     : Starting monit daemon
[SAST Aug 14 17:26:09] info     : Monit start delay set -- pause for 300s
[SAST Aug 14 17:33:08] info     : 'system_MusicBox' Monit started