pimusicbox/pimusicbox #287: [Question] About the Pi-MusicBox

gamecompiler, 2015-05-30:
Hi, 

If id understand your sourcefiles correctly, 
i could "port" your MusicBox to any device running linux and which have the correct dependencies installed. If im correct Pi-MusicBox is just a collection of config files.

Please correct me if im wrong !
I would like to use Pi-MusicBox on my Wandboard dual.

Thanks in advance.

kingosticks, 2015-05-30:
Pretty much. There are some specific bits and optimisations but if you were
prepared to go through it then it's doable. Sometime soon we'll move to a
generic musicbox package which will make this easy. I keep saying soon but
I do mean it.

kingosticks, 2015-06-21:
If you have any specific questions on the subject please reopen this issue.

closed 2015-06-21 by kingosticks