pimusicbox/pimusicbox #291: No advertisement for Mopidy websocket interface

kingosticks, 2015-06-12:
We currently disable Mopidy's built-in zeroconf support and advertise all the musicbox services ourselves. In doing so we have omitted the '_mopidy-http._tcp.local' interface which was added later and which is used by some clients (e.g. Mopidy-Mobile) to discover Mopidy's websocket. We need to add this.

kingosticks, 2015-06-19:
Remotedy also uses this interface.

kingosticks, 2015-06-24:
@tkem I was playing around with this and now I can see `_mopidy-http._tcp` in ZeroConf Browser running on my Android phone but mopidy-mobile can't see it (and neither can remotedy). Is there any extra magic it's looking for in the service advert?

kingosticks, 2015-06-24:
ahh, nevermind, turns out the "service on $host" cannot be "service on $host.local" for some reason. Don't understand why to be honest.

tkem, 2015-06-25:
@kingosticks: beats me; but MoMo is just using a Cordova ZeroConf plugin, and the inner workings if that are just "magic" to me ;-)

commit bbce9f38 (2015-08-23): Advertise Mopidy's websocket interface (Fixes #291)
closed 2015-08-23 by kingosticks