pimusicbox/pimusicbox #293: support duriosound is2 dac

remkoboschker, 2015-06-21:
Hi,

I would like to add support for duriosound boards. They provide a kernel module and overlay as well as an image for a barebones mpd install over at http://www.duriosound.org/phpbb/viewtopic.php?f=3&t=7#p39. I would like to use pi-musicbox with this board. I have tried to load the module and apply the overlay, but the settings get overwritten by pi-musicbox's init script. If I set te box to hifiberry dac, I get sound but also some artifacts.

Please advise how I could add support.

Thanks,

Remko

kingosticks, 2015-06-21:
We don't support device tree overlays, they are disabled. If the module is
included in raspbian then just modify setsound to support it. If the module
is not included in raspbian then we probably won't support it.

remkoboschker, 2015-06-22:
Alright, thank you

closed 2015-06-22 by remkoboschker