pimusicbox/pimusicbox #296: Wifi Problems Raspberry Pi2

LarsGermany, 2015-06-29:
Hello,

on the first start MusicBox worked finde. But next day, without any changes, I get this message: 

wlan0: ERROR while getting interface flags: No such device
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan0 : No such device.
wpa_supplicant: /sbin/wpa_supplicant daemon failed to start
run-parts: /etc/network/if-pre-up.d/wpasupplicant exited with return code 1
Failed to connect to wpa_supplicant - wpa_ctrl_open: No such file or directory
wpa_supplicant: /sbin/wpa_cli daemon failed to start
run-parts: /etc/network/if-up.d/wpasupplicant exited with return code 1

kingosticks, 2015-06-30:
My guess is that /etc/wpa.conf file is somehow messed up. Do you by chance have a '\' character in your password? Or something other than `0-9a-Z_` ? Could you try editing the wireless password field in settings.ini by hand (as opposed to using websettings) and then reboot ?

LarsGermany, 2015-06-30:
No, there are only normal letters. Editing don´t help. I tried with a new installation on a second SD-card, but this installation show the same message after the first boot!

I will tried with an other wifi dongle.

LarsGermany, 2015-06-30:
It works with the other wifi dongle!

closed 2015-06-30 by LarsGermany

kingosticks, 2015-06-30:
Do you know what type the bad one was, then I can look out for it I future
issues.