pimusicbox/pimusicbox #298: Scrobling time is way off

liftarn, 2015-07-03:
Yesterday I listened to some tracks on the MusicBox and then today with Clementine on my computer. I notice the scrobble time is way off on the MusicBox (several hours into the future).

kingosticks, 2015-07-03:
Is the system time correct? The correct timezone?

liftarn, 2015-07-04:
It wasn't so I set it like:
```
root@MusicBox:~# date
Sat Jul  4 09:24:10 CEST 2015
```
And now the scrobbles seem to disappear entirely. I get "now playing", but I don't know if they just vanish or what happens to played tracks.

liftarn, 2015-07-04:
Correction. After cold reboot it seems to be working and the time is only slightly off.
I can live with that.

bonelifer, 2015-07-04:
Is NTP setup on your system?  http://www.cyberciti.biz/faq/debian-ubuntu-linux-install-ntpd/

kingosticks, 2015-07-04:
Ntp should update as part of the boot process. I think.

liftarn, 2015-07-05:
Yes, booted up today. Time is set right, but no scrobbles.

kingosticks, 2015-07-05:
Have a look for any errors in /var/log/mopidy/mopidy.log

liftarn, 2015-07-05:
Correction: I get *some* scrobbles.
I found a few 
2015-07-05 10:30:57,892 - WARNING  Error submitting playing track to Last.fm: NetworkError: [Errno 110] Connection timed out
that may explain it. Streaming music (from Spotify) works so it's not a general network error.

liftarn, 2015-08-25:
On and off the time is still way off (8-10 hours). The time zone is correct. I suspect NTP fails in some way.
The scrobbling problems seem to be unrelated.