pimusicbox/pimusicbox #300: Zeroconf domain name (musicbox.local) no longer resolving

infovore, 2015-07-14:
So this is strange: after a long while of working on my home network (plugged into router via ethernet), http://musicbox.local no longer resolves. I can both SSH in and use the MPD interface _if_ I go to the IP address of the musicbox; but I can no longer use the zeroconf domain name.

As an added side effect, this means that the Airplay functionality is no longer working.

Any ideas where I start looking? (My musicbox is unmodified).

jcass77, 2015-07-19:
This happens to me all the time. I haven't really looked into it, but running ```service avahi-daemon restart``` seems to get it working again.

This is only a temporary fix however, it always goes down again after a short while.

infovore, 2015-07-19:
I had no luck with restarting `avahi-daemon`; however, rebooting my router helped.

liftarn, 2015-07-20:
I never got it to work so I just use the IP.

brimur, 2016-01-21:
For those that come across this thread. Just update the file below with the text below and all the clients will pick it up...

root@MusicBox:~# cat /etc/avahi/services/musicbox.service

<?xml version="1.0" standalone='no'?>
<!DOCTYPE service-group SYSTEM "avahi-service.dtd">
<service-group>
  <name replace-wildcards="yes">MusicBox on %h</name>
  <service>
    <type>_http._tcp</type>
    <port>80</port>
  </service>
  <service>
    <type>_mopidy-http._tcp</type>
    <port>80</port>
  </service>
</service-group>
root@MusicBox:~#

brimur, 2016-01-21:
For those you come across this thread just update your musicbox.service with the text below and reboot. All the mobile apps will find it then

```
root@MusicBox:~# cat /etc/avahi/services/musicbox.service

<?xml version="1.0" standalone='no'?>
<!DOCTYPE service-group SYSTEM "avahi-service.dtd">
<service-group>
  <name replace-wildcards="yes">MusicBox on %h</name>
  <service>
    <type>_http._tcp</type>
    <port>80</port>
  </service>
  <service>
    <type>_mopidy-http._tcp</type>
    <port>80</port>
  </service>
</service-group>

root@MusicBox:~#
```