pimusicbox/pimusicbox #303: Undocumented Realtek 8188EU WiFi driver in Pi MusicBox 0.6

DavisNT, 2015-07-20:
The building of `/lib/modules/3.18.7+/kernel/drivers/net/wireless/8188eu.ko` that is contained within Pi MusicBox 0.6 image (file `musicbox0.6.img`) as well as origin of its sources cannot be found anywhere in GitHub ( https://github.com/woutervanwijk/Pi-MusicBox ). Within Pi MusicBox 0.6 image there is a file named `/etc/modprobe.d/8188eu.conf` that contains:

    # r8188eu is staging, 8188eu is off-kernel
    blacklist r8188eu
    #blacklist 8188eu
    #options r8188eu rtw_power_mgnt=0 rtw_enusbss=0
    options 8188eu rtw_power_mgnt=0 rtw_enusbss=0 

However these contents cannot be found in any revision of this file in GitHub ( https://github.com/woutervanwijk/Pi-MusicBox/blob/develop/filechanges/etc/modprobe.d/8188eu.conf ). 

Moreover the `8188eu.conf` (version found within Pi MusicBox 0.6 image) blacklists (disables) Raspbian built-in Realtek 8188EU breaking Realtek 8188EU support upon kernel update.
Why there is the custom driver preferred (and why it is not documented anywhere)? From where does this driver come? And is it really better than Raspbian default driver?!

And (if it really is better than the Raspbian's driver) shouldn't it be upstreamed to Raspbian?

kingosticks, 2015-07-22:
I think @woutervanwijk can answer this best. But from the comments and a quick search it looks like r8188eu was only included in Raspbian at the start of the year. I'm guessing that in the v0.6 release, this new r8188eu driver conflicted with the historical 8188eu driver we'd used so r8188eu was just disabled to have everything working the same as in older releases. If that is the case, then going forward we should indeed use r8188eu.

kingosticks, 2015-09-14:
Does anyone have a TL-WN725N wifi stick they can test this with if we remove the custom driver?

kingosticks, 2015-09-14:
Urghh it seems this was more complicated than what I wrote before.
* r8188eu.ko is included in Raspbian Wheezy but requires the separate firmware rtl8188eufw.bin driver file which is not. But it is available in firmware-realtek for [wheezy-backports](https://packages.debian.org/wheezy-backports/firmware-realtek) and [jessie](https://packages.debian.org/jessie/firmware-realtek).
* Some people have been building a 8188eu.ko driver themselves and blacklisting r8188eu.ko but I can't work out why, or where they're getting the rtl8188eufw.bin firmware from when they do so.  It also appears that you then need a different driver depending on if you using a v2 or v1 Raspberry Pi so I would assume that what we have in Musicbox v0.6 only works for one or the other.
* I think I need to purchase one of these.

kingosticks, 2015-09-14:
Some relevant links on the subject:
* Raspbian IRC log: http://pastebin.com/qJjT9GHN
* https://github.com/raspberrypi/linux/issues/764
* Random distributions of 8188eu.ko :/  https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=62371&sid=bbbc77afd690d3e6ec6b5f98fbd00a50

DavisNT, 2015-09-15:
I have TL-WN725N (AFAIK v2) and it is working great with the built-in WiFi driver! It was working with latest Raspbian kernel/drivers as well as with kernel/drivers installed by `rpi-update`.

I assume the custom driver is some relic from times when EU8188 driver was not included is Raspbian. See "NOTE" on http://blog.samat.org/2014/12/15/realtek-8188eu-based-wi-fi-adapters-on-the-raspberry-pi/ for more details.

kingosticks, 2015-09-15:
Oh excellent. It must be v2 if you are using 8188eu (v1 uses something
completely different). It doesn't really make sense considering what I was
reading in those links but it makes things much easier, so thanks for
clarifying. As it happens I did just compile the updated v0.43 of
firmware-realtek from wheeezy-backports, nevermind.

kingosticks, 2015-09-15:
Can I double check, you are running wheezy and not jessie, correct?

DavisNT, 2015-09-15:
> Can I double check, you are running wheezy and not jessie, correct?
Yes, I am running Wheezy.

kingosticks, 2015-09-16:
Some Raspbian users have [complained about instability with the included r8188eu driver](https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=113753#p778347). Have you had any problems at all? It's possible these people are not using good enough power supplies as some wireless sticks are very hungry.

The above link has at least confirmed that the 8188eu.ko driver which people have been compiling does include the firmware blob rather than having it as a separate file. As of his morning I am now a proud owner of a TL-WN725N v2 stick so I can do some extra testing.

sonataopus111, 2015-09-16:
Thanks a lot!  I also have a TL-WN725N that did not work with MusicBox 6.0
and I was not able to fix it with the driver files I found.

It did work however with the latest OSMC distribution on newer Raspian
version jessie. I did not have more time to try, and now you have solved
it, thank you!

Pi Music Box with Spotify Premium (at least on "classical" music catalogue)
and other services is an even greater revolution than the introduction of
the CD was in the 80's!

Congratulations

kingosticks, 2015-09-16:
I did some testing earlier with my shiny new TL-WN725N v2 stick, results as follows:

1. With latest updated Raspbian Wheezy - doesn't work and complains the required firmware blob is missing:
> [   12.719736] r8188eu 1-1.4:1.0: Direct firmware load for rtlwifi/rtl8188eufw.bin failed with error -2
[   12.719769] r8188eu 1-1.4:1.0: Firmware rtlwifi/rtl8188eufw.bin not available
[   12.719787] MAC Address = c4:e9:84:1b:85:e4
[   16.874949] usb 1-1.4: USB disconnect, device number 4
[   16.875491] R8188EU: ERROR indicate disassoc

2. As 1. but also installed firmware-realtek version from Wheezy - same problem.
3. As 1. but instead installed firmware-realtek v0.43 from Jessie - works!

@DavisNT Maybe if you are interested you could double check you really have a v2 dongle. Raspbian obviously doesn't not include firmware blobs in their kernel so maybe you're using a different driver, presumably because you've got v1 hardware? It's not a big deal but I am curious. I think we'll try to include the Jessie version of firmware-realtek if the next Musicbox release is still based on Wheezy.

closed 2015-09-16 by kingosticks

DavisNT, 2015-09-17:
I will do `lsusb` and post output here a bit later today, however I am pretty sure that I have something with 8188EU chipset.

I had WiFi broken when doing kernel update on Musicbox 0.6, solution was to un-blacklist the Raspbian official driver (I remember that kernel update did **not** broke WiFi on Musicbox 0.53). 
I will check whether the blob is present on Musicbox 0.6 and try to find out whether it was installed by some Raspbian apt package or otherwise.

DavisNT, 2015-09-17:
As shown my TL-WN725N has Realtek 8188EU shipset:

    root@Player:~# lsusb
    Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
    Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
    Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
    Bus 001 Device 004: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter

Moreover, the file `/lib/firmware/rtlwifi/rtl8188eufw.bin` (SHA-256 1241ddbfc87f0495e0bf09d8b94c94680b60a3d8eeab89462b3c4d8d3e8a1ee0) I have comes from Wheezly `firmware-realtek` package. I have reinstalled the package and done these commands:

    root@Player:~# sha256sum /lib/firmware/rtlwifi/rtl8188eufw.bin
    1241ddbfc87f0495e0bf09d8b94c94680b60a3d8eeab89462b3c4d8d3e8a1ee0  /lib/firmware/rtlwifi/rtl8188eufw.bin
    root@Player:~# apt-file search rtl8188eufw.bin
    firmware-realtek: /lib/firmware/rtlwifi/rtl8188eufw.bin
    root@Player:~# apt-cache policy firmware-realtek
    firmware-realtek:
      Installed: 0.43~bpo8+rpi2
      Candidate: 0.43~bpo8+rpi2
      Version table:
     *** 0.43~bpo8+rpi2 0
            500 http://archive.raspberrypi.org/debian/ wheezy/main armhf Packages
            100 /var/lib/dpkg/status
         0.36+wheezy.1 0
            500 http://mirrordirector.raspbian.org/raspbian/ wheezy/non-free armhf Packages
    root@Player:~#

@kingosticks Are you really sure there was no human error when you tried Wheezly `firmware-realtek` package?
> As 1. but also installed firmware-realtek version from Wheezy - same problem.

kingosticks, 2015-09-17:
OK, this makes makes sense now. You are using v0.43 Wheezy backport courtesy of raspberrypi.org (note, not Raspbian) which was uploaded July 2015. My system isn't using archive.raspberrypi.org (usually defined in /etc/apt/sources.list.d/raspi.list) but I've no idea how I've managed that. 

It certainly explains why I don't personally see the fixed v0.43 package and it's great news that it's available without jumping through hoops like I was doing. It seems a bit unusual but I guess they had no choice but to provide it for Wheezy since they've still not got a Jessie release. 

Thanks for help making sense of this.

commit 1234dd91 (2015-09-20): Use RPF's backported r8188eu wireless driver. Fixes #303