pimusicbox/pimusicbox #308: Create Debian packages for Mopidy extensions

kingosticks, 2015-07-25:
To aid the packaging effort:
* mopidy-local-sqlite
* mopidy-youtube
* mopidy-podcast, mopidy-podcast-itunes, mopidy-podcast-gpodder.net
* mopidy-tunein
* mopidy-spotify-tunigo - is this sticking around?
* webclients:
  * mopidy-mobile
  * mopidy-moped
  * mopidy-mopify
* mopidy-subsonic - not sure if this is maintained..

jodal, 2015-07-25:
I'll see what I can do about this :-)

jodal, 2015-07-31:
Now available at apt.mopidy.com, and hopefully soon in Debian:

- mopidy-local-sqlite
- mopidy-youtube
- mopidy-tunein

jodal, 2015-07-31:
I can't imagine getting the web clients packaged and into Debian because that would require the Debian package to be able to rebuild the JS parts through the use of JS packaged in Debian instead of using npm. Though, we can make debs of the webclient and keep them just at apt.mopidy.com.

kingosticks, 2015-07-31:
This is fantastic. Thanks so much. I would be more than happy if they were
just on apt.mopidy.com.

jcass77, 2015-08-02:
Is mopidy-local-images a candidate for inclusion in the standard distribution perhaps?

kingosticks, 2015-08-02:
it probably is. I think I saw on IRC that @tkem mentioned moving it under https://github.com/mopidy along with his other excellent projects.

tkem, 2015-08-02:
I think it would make sense to at least move mopidy-local-sqlite to the mopidy org repo, especially since @jodal already maintains the Debian package. I also have no objections to moving mopidy-local-images there, if that makes things easier, e.g. if/when the local back end gets split from core and I'm not available to adjust dependencies.

jodal, 2015-08-03:
I guess mopidy-local-images is a candidate for Debian packaging too.

@tkem Feel free to mvoe mopidy-local-sqlite and mopidy-local-images to the mopidy org. You should have access to do so. Please add "jodal" and "adamcik" to the packages at PyPI too, so we're able to make releases if needed.

tkem, 2015-08-03:
@jodal: Great, will do so in a couple of days when I get back from summer vacation.

jodal, 2015-08-04:
Now available at apt.mopidy.com, and hopefully soon in Debian:

- mopidy-podcast
- mopidy-podcast-itunes
- mopidy-podcast-gpodder

jodal, 2015-08-04:
@kingosticks You should change that original list to a checklist we can tick off on ;-)

jodal, 2015-08-12:
Come to think of it mopidy-spotify-tunigo already exists on apt.mopidy.com. Another one to tick off.

kingosticks, 2015-08-13:
I got this error just now when trying to install mopidy-youtube from apt on raspbian:
```
The following packages have unmet dependencies:
 mopidy-youtube : Depends: python-pafy but it is not installable
E: Unable to correct problems, you have held broken packages. 
```
And then:
```
# apt-cache show python-pafy
N: Can't select versions from package 'python-pafy' as it is purely virtual
N: No packages found
```

Does that make any sense?

jodal, 2015-08-13:
Yes, see https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=795007 and https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=795085

PanderMusubi, 2015-08-24:
Nice work. If this covers all of https://github.com/mopidy/mopidy-youtube/issues/44 then that issue can be closed. Thanks.

jodal, 2015-08-25:
Copying the answer here for completeness:

> I've recently packaged mopidy-youtube and it is available on apt.mopidy.com as well as in Debian unstable, testing and Ubuntu 15.10. The problem is, the package is broken because the dependency python-pafy has removed the Python 2.7 support from its package. This is reported as a bug at https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=795085, but I've received no response so far.

elsurudo, 2016-01-15:
Is there any workaround for the "pafy" issue for Raspbian Wheezy currently? Unfortunately, I can't install "mopidy-youtube" currently.

kingosticks, 2016-01-15:
You should be able to install it from pip.

elsurudo, 2016-01-15:
Thanks, that did the trick indeed.

jodal, 2016-01-25:
Now available at apt.mopidy.com, and hopefully soon in Debian:

- mopidy-somafm
- mopidy-internetarchive

kingosticks, 2016-02-06:
Added mopidy-gmusic. Maybe there is a reason this was left off the list but I can't find that reason anywhere now.

PanderMusubi, 2016-02-07:
Can the issue https://github.com/AlexandrePTJ/mopidy-somafm/issues/22 be closed please? Thanks

jodal, 2016-02-07:
@kingosticks Can't see mopidy-gmusic anywhere. Where did you add it? Maybe include the other packages I've made in the original list too?

@PanderMusubi Why are you asking here for that issue to be closed? I opened it for @AlexandrePTJ to know that there is now a Debian package of his extension and that he might want to include a notice in his README. It is up to him to do so or not and to close the issue.

PanderMusubi, 2016-02-07:
@jodal OK, but as it is in the repo I thought it can be closed.

jodal, 2016-02-07:
@PanderMusubi Well, no, the issue is about adding a notice in the readme, not about making the Debian package.

kingosticks, 2016-02-08:
@jodal I must have added it to the OP and then forgotten to click the update button. That, or it was a particularly exciting dream. Sorry for the confusion there. That list should be up to date now, although I have not verified all the versions are current.

jodal, 2016-02-08:
My Debian QA tracker page gives me a quick overview of what packages are out of date compared to PyPI. The only stuff not listed there is python-spotify, mopidy-spotify, python-tunigo and mopidy-spotify-tunigo. The last two where updated yesterday, and are currently the only packages I know of that are not up to date on apt.mopidy.com.

As for other updates to the list:

- mopidy-spotify-tunigo seems to be sticking around.
- mopidy-subsonic has no release for two years, and the only change the last year is the merge of my PR to support playback under Mopidy 1.0.
- mopidy-local-images just needs to be packaged.
- mopidy-gmusic somewhat surprisingly only needs gspoauth and gmusicapi to be packaged first. All other deps seems to be in Debian in acceptable versions.
- The web clients I haven't looked at. Probably, none of them will be easy to get into Debian proper as they probably depend on various JS build systems and JS libs not packaged in Debian. Making apt.mopidy.com-only packages of the PyPI releases (where the JS building stuff is already done for you) should be straight forward, though.